pybamm-team/PyBaMM #906: Comparison class

Scottmar93, 2020-03-22:
I might be useful to provide a BatchStudy class or adapt the Simulation class to make it easier to compare models / discretizations / solvers / run batches of experiments and generate statistics for solvers. Usage would be something like: 

```python
batch = pybamm.BatchStudy(models={"SPM": spm, "DFN": dfn}, solvers={"casadi": casadi, "klu": klu"}, experiments={"constant current": constant_current, "cccv": cccv}, repeats=10)  #repeat solves to generate speed statistics for solvers

batch.solve()
batch.save("batch_study.p")
```

```python
batch = pickle.load("batch_study.p")
batch.plot(models=["SPM"], experiments=["cccv"], variables=["Terminal voltage [V]"])
batch.solver_stats(models=["SPM"], experiments=["constant current"]) # no argument compares all solvers
batch.export_data(models=["SPM", "DFN"], experiments=["constant current"], variables=["Terminal voltage [V]"])
```

Would be best to ensure that the class is picklable so that people can basically just batch run a whole bunch of different experiments etc on a bunch of models and solvers and then load up one object and play around with plotting / solving statistics.

tinosulzer, 2021-03-10:
When you want to compare different models, experiments, etc, you have to use a for loop, e.g.
```python3
# load models
models = [pybamm.lithium_ion.SPM(), pybamm.lithium_ion.SPMe(), pybamm.lithium_ion.DFN()]

# create and run simulations
sims = []
for model in models:
    sim = pybamm.Simulation(model)
    sim.solve([0, 3600])
    sims.append(sim)

# plot
pybamm.dynamic_plot(sims)
```
or
```python3
# load solvers
model = pybamm.lithium_ion.DFN()
solvers = [pybamm.CasadiSolver(mode="fast"), pybamm.CasadiSolver(mode="safe")]
# create and run simulations
sols = []
sim = pybamm.Simulation(model)
for solver in solvers:
    sol = sim.solve([0, 3600], solver=solver)
    sols.append(sol)

# plot
pybamm.dynamic_plot(sols)
```
The idea is instead to have a class you can call that you pass the lists to and it does the for loop for you

priyanshuone6, 2021-03-17:
```
batch = pybamm.BatchStudy(models={"SPM": spm, "DFN": dfn}, solvers={"casadi": casadi, "klu": klu"}, experiments={"constant current": constant_current, "cccv": cccv}, repeats=10)  #repeat solves to generate speed statistics for solvers

```
Like in the above example, is it like the first model will run for all the listed solvers and for all the listed experiments or is it like the first model will use the first solver and the first experiment and the same goes for the second model.
Also, what do the `repeats` signify?

tinosulzer, 2021-03-25:
Might be good to have both, and a `permutations` True/False keyword arg to decide which to use. Default should be False (first model with first solver, second model with second solver, etc) IMO. 

The `repeats` is about how many times "solve" should be called. The result will be the same each time but calling "solve" several times can be good to get a better estimate of the solve time (which can vary from solve to solve depending on what else your computer is currently trying to do)

PR #1455 (2021-04-09): add BatchStudy class
# Description

Add class to compare different models and conditions

Fixes #906 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ [... cut]
closed 2021-04-09 by tinosulzer