pybamm-team/PyBaMM #908: Porting the CI/CD system to GitHub Actions

dalonsoa, 2020-03-23:
PR #855 started to implement the current CI system in GitHub Actions. A couple of things are missing to make it fully functional and be able to replace Travis and Appveyor:

- [ ] Solve problem with Sundials libraries in MacOS. They seem to install properly, but tests fail, complaining they are not found. 
- [ ] Implement a scheduled testing. This should be easy by using [`on.schedule`](https://help.github.com/en/actions/reference/workflow-syntax-for-github-actions#onschedule). 
- [ ] Implement a CD system to deploy PyBaMM to PyPI after a tagged commit to the master branch. Again, this should be easy using [this Action](https://github.com/pypa/gh-action-pypi-publish).

tinosulzer, 2020-03-23:
I also had problems with sundials libraries in MacOS. One difference is you need to specify `DYLD_LIBRARY_PATH` instead of `LD_LIBRARY_PATH`.

tinosulzer, 2020-05-18:
I think this can be closed? Re-open if not

closed 2020-05-18 by tinosulzer