pybamm-team/PyBaMM #909: 1 sec output dataset

gyouhoc, 2020-03-23:
I want to have 1sec or 0.1 sec time series output dataset. 
Is there a good way to produce these dataset? 

1. when I have a normal input current (constant)?

2. when I have a current profile csv file? Also, would you direct me to the place where you explain how I load a current profile csv file to your model? I cannot find the location you explain this.   
 

**To Reproduce**
sim = pybamm.Simulation(model)
sim.solve(t_eval=np.linspace(0, 5,num=18000))

sim.solution.save_data(
    "outputs_20200318_1000_1.csv",
    [
        "Time [s]",
        "Terminal voltage [V]",
        "Current [A]",
    ],
    to_format="csv",
)   

**Expected behaviour**
1sec or 0.1sec time series data

**Screenshots**

[image]

tinosulzer, 2020-03-23:
Make sure you have updated to the latest version (at least v2 release, or master or develop versions) as we recently changed the interface for this.
For time:
```python3
t_eval=np.arange(0, 3600, 1) # 1 second timesteps
t_eval=np.arange(0, 3600, 0.1) # 0.1 second timesteps
```
For currents: see [this example](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/change-input-current.ipynb) (look for "US06") and you will need to add your current cycle to the file "inputs/drive_cycles" as it looks there for drive cycles

tinosulzer, 2020-03-25:
@gyouhoc please let me know if this solves your issue and I will close this (or you can close and comment yourself)

gyouhoc, 2020-03-25:
I have this error message, "SolverError: Maximum number of decreased steps occurred at t=0.125. Try solving the model up to this time only." I guess this is due to the old version of my pybamm software. I will reinstall pybamm and I will try again. Let you know.

gyouhoc, 2020-03-25:
This is working well thx

closed 2020-03-25 by gyouhoc