pybamm-team/PyBaMM #910: make a fixed parameter to a variable

gyouhoc, 2020-03-24:
**Summary**

In the version of the software that I have in the default setting, "D_ref" for electrodes is a fixed parameter. Would you make this as a variable so that I can tweak this number?  

 **Motivation**
 Why are we doing this? What use cases does it support? What is the expected outcome?

I want to set the parameters so that I have better fitting in high current or low temp applications (no test yet but it is my expectation).  

**Additional context**
 Add any other context or screenshots about the feature request here.

[image]

tinosulzer, 2020-03-24:
You can either change the parameter manually:
```python3
model = pybamm.lithium_ion.DFN()

param = model.default_parameter_values

def diffusivity(sto, T, T_inf, E_D_s, R_g):
    D_ref = 3.9 * 10 ** (-14)
    arrhenius = pybamm.exp(E_D_s / R_g * (1 / T_inf - 1 / T))

    return D_ref * arrhenius

param["Negative electrode diffusivity [m2.s-1]"] = diffusivity

sim = pybamm.Simulation(model, parameter_values=param)
sim.solve()
sim.plot()
```
or make it an input and then specify it when you solve:
```python3
model = pybamm.lithium_ion.DFN()

param = model.default_parameter_values

def diffusivity(sto, T, T_inf, E_D_s, R_g):
    D_ref = pybamm.InputParameter("D_ref")
    arrhenius = pybamm.exp(E_D_s / R_g * (1 / T_inf - 1 / T))

    return D_ref * arrhenius

param["Negative electrode diffusivity [m2.s-1]"] = diffusivity

sim = pybamm.Simulation(model, parameter_values=param)
sim.solve(inputs={"D_ref": 3.9 * 10 ** (-12)})
sim.plot()
```

tinosulzer, 2020-03-25:
@gyouhoc please let me know if this solves your issue and I will close this (or you can close and comment yourself)

gyouhoc, 2020-03-25:
Is there another method that I can parameterize D_ref as other parameters? 
So that I can tweak the number with following commend?
parameter_values2["Negative reaction rate activation energy [J.mol-1]"]=6000

tinosulzer, 2020-03-25:
More or less:
```python3
import pybamm

model = pybamm.lithium_ion.DFN()

param = model.default_parameter_values

def diffusivity(sto, T, T_inf, E_D_s, R_g):
    D_ref = pybamm.Parameter("D_ref")
    arrhenius = pybamm.exp(E_D_s / R_g * (1 / T_inf - 1 / T))

    return D_ref * arrhenius

param["Negative electrode diffusivity [m2.s-1]"] = diffusivity

# you can update the value of D_ref on this line
param.update({"D_ref": 3.9 * 10 ** (-14)}, check_already_exists=False)

sim = pybamm.Simulation(model, parameter_values=param)
sim.solve(inputs={"D_ref": 3.9 * 10 ** (-12)})
sim.plot()

```

gyouhoc, 2020-03-25:
Is there anyway that I can list "D_ref" in the default para list?
So that I can update the value as 

model.default_parameter_values["D_ref "]=3.9 * 10 ** (-12)

tinosulzer, 2020-03-25:
```python3
import pybamm

model = pybamm.lithium_ion.DFN()

param = model.default_parameter_values

def diffusivity(sto, T, T_inf, E_D_s, R_g):
    D_ref = pybamm.Parameter("D_ref")
    arrhenius = pybamm.exp(E_D_s / R_g * (1 / T_inf - 1 / T))

    return D_ref * arrhenius

param["Negative electrode diffusivity [m2.s-1]"] = diffusivity
# Do this first
param.update({"D_ref": 3.9 * 10 ** (-14)}, check_already_exists=False)

# then you can update the value of D_ref on this line
param["D_ref"] = 3.9 * 10 ** (-12)

sim = pybamm.Simulation(model, parameter_values=param)
sim.solve(inputs={"D_ref": 3.9 * 10 ** (-12)})
sim.plot()
```

gyouhoc, 2020-03-25:
can this do the same thing? 

sim = pybamm.Simulation(model, parameter_values=param)
sim.solve() <= removing the input definition  
sim.plot()

tinosulzer, 2020-03-26:
oops, yes, that was a typo, removing it does the same thing

gyouhoc, 2020-03-26:
Np. Thank you for your help.

closed 2020-03-26 by gyouhoc