pybamm-team/PyBaMM #950: Add get_value method to simulation

Scottmar93, 2020-04-08:
Add a method to simulation that accepts either a string: "Negative electrode thickness [m]" or a symbol `l_n` and returns the current value of that parameter. Basically something like: 

```python
def get_value(symbol):
     return self.parameter_values.process_symbol(symbol).evaluate()
```

with some more checks to ensure the parameter can be evaluated etc. This could also also be extended for `FunctionParameters` to evaluate at particular value of an input.

tinosulzer, 2020-04-08:
By the way, you can already do this using the parameter values class:
```python3
parameter_values.evaluate(symbol)
```

tinosulzer, 2020-06-15:
Closing as I think `sim.parameter_values.evaluate(symbol)` is clear and concise enough. Reopen if not.

closed 2020-06-15 by tinosulzer