pybamm-team/PyBaMM #955: Rest preceding discharge step causes DAE solver bug (Chen 2020)

Kieran-ORegan, 2020-04-13:
Including a rest before the following discharge step does not allow successful simulation due to a bug with 'DAE solver'.
 
"Newton/Linesearch algorithm failed to converge"

experiment = pybamm.Experiment(
                     ["Rest for 5 minutes",
                         "Discharge at 0.5C until 2.5 V",
            "Rest for 120 minutes"],
           period="10 seconds",
)
model = pybamm.lithium_ion.DFN()
chemistry = pybamm.parameter_sets.Chen2020
parameter_values = pybamm.ParameterValues(chemistry=chemistry)
parameter_values["Negative electrode diffusivity [m2.s-1]"] = 1.3e-14
sim = pybamm.Simulation(model, parameter_values=parameter_values, experiment=experiment)
sim.solve()
 
[image]

rtimms, 2020-04-13:
looks like this will be fixed by PR #951

tinosulzer, 2020-04-13:
Yep this should be fixed now

closed 2020-04-13 by tinosulzer