pybamm-team/PyBaMM #960: Loading simulations

Scottmar93, 2020-04-15:
When loading a simulation using `sim.load("something.p")` I get the error:
```
No module named 'graphite_diffusivity_Kim2011'
```
But if I add the lines:
```
chemistry = pybamm.parameter_sets.NCA_Kim2011
parameter_values = pybamm.ParameterValues(chemistry=chemistry)
```
before loading the simulation, it loads and works fine. 

Can anyone else can reproduce this? Might just be an issue with my own python setup.

rtimms, 2020-04-15:
yeah I get the same. looks like #792 is related.

also note that if you try to save after plotting you get 
```  
File "/home/user/Documents/PyBaMM/pybamm/simulation.py", line 748, in save
    pickle.dump(self, f, pickle.HIGHEST_PROTOCOL)
TypeError: can't pickle _tkinter.tkapp objects
```
might be worth fixing this too (presumably you can just remove all the plot stuff before saving)

tinosulzer, 2020-04-15:
Yeah I get this issue too. Also get it if trying to save a simulation where one of the parameter functions has been defined locally.
It's definitely a problem with loading modules but I have no idea how to fix it.

tinosulzer, 2020-04-15:
It looks like the problem is that the path to a particular function (e.g. graphite_diffusivity_Kim2011) is not added to the sys path by default. Some possible workarounds:

1. (preferred but can't find how to do this) redefine the function locally inside `ParameterValues`. If not possible with python then maybe we could do it by creating some expression tree.
2. Automatically import functions defined in inputs/parameters to the pybamm package. This isn't ideal as it doesn't deal with the case where a function has been defined locally
3. Use something other than pickle that doesn't have this issue

priyanshuone6, 2021-04-16:
@tlestang Are you working on this issue? Can I work on this

tlestang, 2021-04-22:
No I'm not, feel free to work on this

PR #1475 (2021-05-05): Make parameters importable
# Description

- Modify load_function as it imports the function without the complete module path and that causes issues while unpickling. It also removes the (incomplete) module path from `sys.modules` to avoid namespace clashes.
- Edit tests
- Add `__init__.py` in parameters to convert them to modules and make them importable through pybamm

Fixes #960

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If [... cut]
closed 2021-05-05 by tlestang