pybamm-team/PyBaMM #966: exchange current density

gyouhoc, 2020-04-20:
Hi, 

As I review the list of the input parameters for both anode and cathode, I realize that there is no parameter for the exchange current density ,io, for both electrodes. Would you direct me how to setup this number for the simulation? 

thank you

tinosulzer, 2020-04-20:
The relevant parameter is "reaction rate", then the exchange-current density is hard-coded in as 
```python3
i0 = reaction_rate * (
                c_e ** (1 / 2) * c_s_surf ** (1 / 2) * (1 - c_s_surf) ** (1 / 2)
            )
```
It would probably be more standard to have an exchange-current density parameter though (which takes in c_s_surf and c_e), right @rtimms @Scottmar93 @ferranbrosa @TomTranter ?

rtimms, 2020-04-20:
will need to take `c_s_max` etc as it will need to be defined in dimensional form to be consistent with other parameters

PR #975 (2020-05-01): Issue 966 exchange current
# Description

Make "exchange-current density" a parameter instead of "reaction rate".
Also replaced "surface area density" with "surface area to volume ratio"

Fixes #966 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x [... cut]
closed 2020-05-01 by tinosulzer