pybamm-team/PyBaMM #972: adding arctan function in pybamm

gyouhoc, 2020-04-29:
Hi, 

I have a OCP function as 

    u_eq = (
        + 0.1493 + 0.8493*exp(-61.79*sto)
        + 0.3824*exp(-665.8*sto) - exp(-39.42*sto - 41.92)
	- 0.03131*atan(25.59*sto - 4.099) 	
	- 0.009434*atan(32.49*sto - 15.74)
	)

And I just realize that there is no atan function defined in pybamm. 
I tried to add an atan function as below but it doesn't work. 
Would you let me know the proper way to add atan function in pybamm? 

In the function file, I add following lines. 

class Arctan(SpecificFunction):
    """ Arctan function """

    def __init__(self, child):
        super().__init__(np.arctan, child)

    def _function_diff(self, children, idx):
        """ See :meth:`pybamm.Function._function_diff()`. """
        return 1 / (children[0] ** 2 + 1)

def atan(child):
    " Returns hyperbolic tan function of child. "
    return pybamm.simplify_if_constant(Arctan(child), keep_domains=True)

However, I still have a following error. 

     18         + 0.3824*exp(-665.8*sto) - exp(-39.42*sto - 41.92)
     19         - 0.03131*atan(25.59*sto - 4.099)
---> 20         - 0.009434*atan(32.49*sto - 15.74)
     21 	)
     22 

NameError: name 'atan' is not defined

tinosulzer, 2020-04-29:
Did you import `atan` in the first line of your ocp function file? (`from pybamm import exp, atan`)

gyouhoc, 2020-04-29:
Yes, I did. 

Below is from my input file. 

from pybamm import exp, atan

def graphite_ocp_NCM2020(sto):
    """
       Graphite Open Circuit Potential (OCP) as a function of the stochiometry [1].

       References
       ----------
       .. [1] Kim, G. H., Smith, K., Lee, K. J., Santhanagopalan, S., & Pesaran, A.
       (2011). Multi-domain modeling of lithium-ion batteries encompassing
       multi-physics in varied length scales. Journal of The Electrochemical
       Society, 158(8), A955-A969.
       """

    u_eq = (
        + 0.1493 + 0.8493*exp(-61.79*sto)
        + 0.3824*exp(-665.8*sto) - exp(-39.42*sto - 41.92)
	- 0.03131*atan(25.59*sto - 4.099) 	
	- 0.009434*atan(32.49*sto - 15.74)
	)

    return u_eq

gyouhoc, 2020-04-29:
Nevermind. It is working after I restart the code. 

thx

closed 2020-04-29 by gyouhoc

tinosulzer, 2020-04-29:
Great! When you're ready, it would be great if you could open a PR adding the atan function to the code