pybamm-team/PyBaMM #976: Casadi extrapolate warning/error

ferranbrosa, 2020-05-01:
When Casadi extrapolates you get errors like
```
At t = 0.300171, , mxstep steps taken before reaching tout.
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:830: Linear solve failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:830: Linear solve failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:830: Linear solve failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:830: Linear solve failed
```
which are hard to trace back to the interpolant. It would be useful to throw a warning/error when this happens which is more insightful.

tinosulzer, 2020-05-01:
It's not possible to do this directly as we don't have control over the casadi objects.
But we could add a new type of event to the model, which automatically gets added when an interpolant is added to the model?

ferranbrosa, 2020-05-01:
You mean an event in the solver?

tinosulzer, 2020-05-01:
Yes, events are stored in the model then used by the solver

tinosulzer, 2020-05-01:
Can you create a branch and add a test where it fails?

ferranbrosa, 2020-09-25:
Looking into this issue at the moment and trying to recall what we discussed. The issue is that the Casadi interpolant throws NaNs when it extrapolates (side note here: the default interpolant behaviour is to extrapolate, is Casadi ignoring it or is it taking `extrapolate=False`?).

Adding events would be an easy fix (we then would know which interpolant broke and on which limit). But what do we want to do then? Stop the simulation and just tell the user that this was the issue? Throw an error? Throw a warning?

PR #1315 (2021-01-04): Issue 976 casadi extrapolate warning
# Description

When an interpolant is created from data, a new type of event (`INTERPOLANT_EXTRAPOLATION`) is added to ensure that we can catch when extrapolation occurs. This is particularly important in CasADI as you get a lot of warnings that are not very insightful.

I haven't managed to remove all these warnings as, before hitting the event and raising the error, the integration step where extrapolation happens takes place. However, I don't know how to get rid of them if we can't catch the CasADI warnings.

Fixes #976

## Type of change

Please add a l [... cut]
closed 2021-01-04 by brosaplanella