pybamm-team/PyBaMM #983: Half-cell model for Li-ion battery

schweini2409, 2020-05-06:
**Summary**
Could a half cell model (vs Li foil) be added along with the already-existed full cell model for Li-ion battery? It could be useful to simulate lab tests as they're frequently built in this configuration.

tinosulzer, 2020-05-06:
I'm not familiar with half-cell models, would this include the separator or just a single electrode?

schweini2409, 2020-05-06:
The only difference compared to the full-cell is on the anode side. The anode is now Li foil. Since the reactions ((Li+) + (e-) -> Li) occur at the surface of the Li foil (no thickness), the micromodel that solves the solid-diffusion in the particles can be removed. A Butler-Volmer equation is needed to represent the reaction and a boundary condition that yields the sum of exchange current jk from BV to equal to the total current I.

tinosulzer, 2020-05-06:
Makes sense, and I assume there is then a flux condition for the electrolyte concentration at the separator/anode interface?

schweini2409, 2020-05-07:
Indeed, there is a no flux condition for the electrolyte at the interface separator/anode (Li foil) just like for the current collector (Ne,n (x=0) = 0).

ferranbrosa, 2020-05-07:
Isn't the condition at the separator-Li foil interface -De d ce / dx = I / F (with diffusion De, lithium ion concentration in the electrolyte ce, applied current density I and Faraday constant F)? Then Ne should be equal to (1-t^+) I / F which I guess it is the contribution of the negative ions.

schweini2409, 2020-05-07:
Isn't there a missing term in what you wrote? (1-t^+) should be on the right side as well?
I think the boundary condition at the li foil is that the flux of the anion is zero. Then the diffusion term of the anion is balanced  by the migration term :
-De d(ce)/d(x) ) = (1-t^+)I/F 
You can find the base equations for half cell model in this paper : 
Thomas K.E., Newman J., Darling R.M. (2002) Mathematical Modeling of Lithium Batteries. In: van Schalkwijk W.A., Scrosati B. (eds) Advances in Lithium-Ion Batteries. Springer, Boston, MA

hzgezi-nims, 2020-06-09:
In a half-cell, it contains an electrode (cathode or anode) as working electrode, a separator, a piece of pure Li foil as the reference electrode and counter electrode, electrolyte, current collector etc. So the potential discussed in a half cell is the voltage gap between the working electrode and the Li foil. In terms of the amount of Li ions resource from this Li foil, it can be considered as excessive. Practically, half-cell is commonly used as the first step: breaking the question into cathode and anode separately. And then based on the capacity, mass balance, to design a full-cell accordingly.

tanhweejien, 2020-06-23:
Yeah, it would be really useful to be able to simulate a half cell. In the following paper the authors model a NMC111 half cell at different electrode thicknesses. I hope the maths in the paper help.

Modeling the effect of electrode thickness with experimental validation
https://www.sciencedirect.com/science/article/pii/S0360544219315361

schweini2409, 2020-06-23:
I do agree with this! The half cell model is well-known and I think it's quite straightforward to implement it and it should be a default model besides the full cell model.

PR #1121 (2020-08-11): Issue 983 half cell
# Description

Basic DFN model for a half-cell. This is an experimental functionality which doesn't support all PyBaMM features (e.g. experiments) and which will be integrated into the submodel structure in the future.

Fixes #983 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)

# Key checkl [... cut]
closed 2020-08-11 by tinosulzer