pybamm-team/PyBaMM #985: Interactive plot

tinosulzer, 2020-05-06:
Add a new plot type where input parameters can easily be varied (e.g. using a text box) to inspect the effect on variables

tinosulzer, 2021-05-18:
@priyanshuone6 @Saransh-cpp could one of you pick this up? basically I am trying to make a plot with a box where you can change a parameter to see how the plot then changes

Saransh-cpp, 2021-05-18:
> @priyanshuone6 @Saransh-cpp could one of you pick this up? basically I am trying to make a plot with a box where you can change a parameter to see how the plot then changes

Looking into it

Saransh-cpp, 2021-05-22:
@tinosulzer I was able to get it up running with some minor changes. For the part where the plot updates after entering the value, do I need to pass the initial `model` and `parameter values` from the `Simulation` class to the `InteractivePlot` class and solve it again with the updated values, or is there a better way?

I was also going through the scripts and the examples and maybe we can now update the "comparing" scripts with `BatchStudy`?

tinosulzer, 2022-12-09:
Not completed but not in scope

closed 2022-12-09 by tinosulzer