pybamm-team/PyBaMM #989: Warning when "extrapolating" fitted parameter functions

ferranbrosa, 2020-05-11:
**Summary**
The parameter functions usually are provided with a range of validity, but this range is included in the parameter sets. Could we throw a warning if the functions are evaluated outside of this range? Not sure if that can be tackled jointly with #976 

 **Motivation**
Throwing a warning would make users more aware of the limitations of the parameters they are using and might help debugging the code.

ferranbrosa, 2020-09-25:
An easy way here would be to add an if statement inside the function that throws a warning when the fitting range is exceeded, but I suspect that would be slow and it would clutter the terminal with warnings. Not sure if we want that.

A better approach could be to add a non-terminating event (maybe new event type) that registers a warning which is printed after solving. The question here is where to define the event and the range limits.

Thoughts?

tinosulzer, 2021-01-10:
Was this fixed by #1315?

brosaplanella, 2021-01-10:
No, needs some extra work as at the moment the warnings are only defined by the interpolants

tinosulzer, 2021-01-11:
Ah I see the difference now. To define the range, I think we need a new class `BoundedFunctionOutput(output, range)`, and when `ParameterValues` sees this class it puts `output` into the expression tree and `range` into the events. This way we don't have some functions returning just one output (the function) and some returning two (function and range). Unless you had something else in mind :)

brosaplanella, 2021-01-11:
I had a similar idea. Wouldn't it be better to define the class with a `call` method (hopefully then it works the same way as with functions) and an attribute for the bounds? Then your function doesn't output the bounds all the time.