pybamm-team/PyBaMM #992: plot pybamm arrays

rtimms, 2020-05-13:
It would be nice to automate the plotting of `pybamm.Array` objects for doing things like the example below. I think just 1D is fine for now. More complicated will usually be of solution entries, so handled via `Quickplot`, but it would be nice to easily plot parameters that are functions. I guess 2D is useful since this depends on T and sto. Thoughts?

```
import pybamm
import numpy as np
import matplotlib.pyplot as plt

param = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Ecker2015)
D_fun = param["Negative electrode diffusivity [m2.s-1]"]
x = pybamm.Array(np.linspace(0,1,100))
D = D_fun(x, 300)
plt.plot(x.entries,D.entries) 
```

tinosulzer, 2020-05-13:
do you mean something to replace the last line with `pybamm.plot(x,D)`?

rtimms, 2020-05-13:
yeah, sorry I wasn't very clear

tinosulzer, 2020-05-13:
Yeah that sounds like a good feature. And you could do `pybamm.plot([x,T], D(x,T))` for 2D.

PR #1008 (2020-05-26): Issue 992 plot arrays
# Description
Added basic plots for 1D and 2D arrays (`pybamm.plot` and `pybamm.plot2D`). I've left these as separate methods, but I guess they could be combined and the appropriate plot called depending on what arrays you pass. On the other hand, maybe it is better to be explicit. 

Also added `pybamm.linspace` and `pybamm.meshgrid` to quickly create some common arrays.

Fixes #992 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note re [... cut]
closed 2020-05-26 by rtimms