pybamm-team/PyBaMM #998: Faraday tech sessions

rtimms, 2020-05-15:
### We have two session in the Faraday Tech Series:

Tues 2nd June 10:00-11:30. Python Battery Mathematical Modelling (PyBaMM) Part 1: Demonstrating the basics of PyBaMM and how you can use it. Register here: PyBaMM, Part 1

Tues 9th June 10:00-11:30. Demonstrating more advanced features of PyBaMM. Register here: PyBaMM, Part 2
In this session we will demonstrate the speed and flexibility of PyBaMM through a series of case studies, followed by a Q&A discussion on how to utilise PyBaMM in your research.

The idea is the first of these will basically be a condensed version the user workshop and the second will be more in depth with some case studies. I think it makes sense to do a new release before this.

### Possible timetable could be:

2nd June
10:00-10:05 - show how to install/point to website 
10:05-10:20 - pybamm overview (explain motivation/core idea. show off functionality and give an outline for this and the next session)
10:20-10:55 - running a simulation, running an experiment, plotting etc.
10:55-11:15 - changing parameters, settings etc.
11:15-11:30 - questions/discussion

3rd June
10:00-10:05 - intro/welcome
10:05-11:00 - case studies (@ferranbrosa @TomTranter ?)
11:00 - 11:15 - workflow for contributing  
11:15-11:30 discussion

I'm happy to do the first session (unless anyone wants to volunteer for parts of it!), but it would be good if some others could be online to help troubleshoot/answer questions live. The plan is to run on teams/slack as before.

### Things to include in release
- update parameters cli so you can grab an entire directly (just so it is easy for users to see e.g. all lithium-ion parameters in their local directory)
- interactive plot (#985) might be nice, but not essential 
- should we add a dimensional "basic" DFN model so that people who aren't familiar with non-dimensionalisation can hack away at it? could even define params locally to show how to truly do thing self-contained, though maybe this is best left as a separate example
- ???

jedgedrudd, 2020-05-15:
non-dimensionalisation is definitely a popular issue

rtimms, 2020-05-15:
> non-dimensionalisation is definitely a popular issue

as in people find it a barrier to entry for understanding/contributing? or debate about which is the "best"/preferred way to scale things?

I think the former is important for us, but I'd like to avoid discussion of the latter

jedgedrudd, 2020-05-15:
> This email from notifications@github.com<mailto:notifications@github.com> originates from outside Imperial. Do not click on links and attachments unless you recognise the sender. If you trust the sender, add them to your safe senders list<https://spam.ic.ac.uk/SpamConsole/Senders.aspx> to disable email stamping for this address.

Agreed – both are interesting to pursue, but for the workshop, an example to show how it works is all we need for now.

rtimms, 2020-05-15:
ok. I think we probbaly don't want to go down the route of showing how the equations are non-dimensionalised in the workshop, just point out that the models in pybamm are dimensionless. Then point out we have the dimensional "basic" there for people to use if they wish. This won't have the flexibility of the submodel structure, but a possible workflow for people learning/contributing who aren't familiar with non-dim is -> follow/edit "basic" model to do what they want -> ask us for help with non-dim after -> ask for help turning their new addition into a submodel which fits into the structure. Hopefully as we get more users there will be more people that can be asked for help at each stage and the burden on us will be smaller.

ferranbrosa, 2020-05-16:
The timetable looks good to me! I think an interesting case study could be how to add a new parameter set. Some people might be working with own parameters or from a paper that is not included and defining them in the main script takes a lot of space. 

For this, I think the update parameters cli would be necessary so we can show how to incorporate the new parameter set. We would also need to have the blank templates available somewhere. Also related to that, are we thinking of changing the format of the parameter sets (as @tinosulzer suggested over Slack)? Is it worth pushing it to have it for the workshop or given that it would only be a change in the template it is OK if we change it afterwards?

rtimms, 2020-05-20:
should we do #920 before this, just so that all user interfaces are in dimensional form?

TomTranter, 2020-06-01:
Hey guys, I'm happy to contribute to the session on the 9th? Is there an outline for what we want to include and for how long to spend on each case study?

rtimms, 2020-06-01:
Thanks @TomTranter . I think if we aim for ~25 mins per case study that would be good (shorter is ok too, I think it it fine to finish early and leave more time for discussion/people to leave early).

I think @ferranbrosa was going to do something related to parameters?

In terms of what to cover I guess it is pretty open. Maybe if you have a good example of using pybamm to do something outside of just running a simulation that would be good (e.g. your rate capability example or similar?). But basically do whatever you think is interesting and goes a little beyond just using the defaults

closed 2020-06-10 by rtimms