pybamm-team/PyBaMM #1006: process space-only variables

rtimms, 2020-05-20:
At the moment `ProcessedVariable` only works if `len(t)>1` (except for the special case of  2D finite elements, which can handle "fixed t" cases). It would be good to set it up so that any variables that are functions of space only can be processed.

tinosulzer, 2020-05-20:
Yeah would be great to fix that for the solutions of the algebraic solvers too. I had been working around it by giving `[0,1]` as the t_eval for variables that depend only on space.

If we can automatically figure out if the variable was only a function of space, I wonder if we still need to raise that error at all? I guess it might then be hard to figure out that the solver has failed (i.e. hit an event too early) though.

rtimms, 2020-05-20:
I think it would be good to remove that error. I think it should (hopefully) be obvious to the user if there is only one entry in `t` when they were expecting more? 

Do we need to know if a variable depends on space? As in, you can pass any `t` to evaluate and it will just return the same thing? I think the name I used for this issue is wrong, and what I really meant was fix for `len(solution.t)=1`, in which case you can still evaluate the variable in the same way, but need to create an interpolant of lower dimension.

tinosulzer, 2020-05-20:
Yes, I think just creating a lower-dimensional interpolant in the case where `len(solution.t)=1` would be appropriate. Maybe the ODE/DAE solvers (i.e. non-algebraic) themselves should raise the error still?

rtimms, 2020-05-22:
Yeah I think that is the best place to raise that error

PR #1020 (2020-05-27): Issue 1006 process space var
# Description
Extend `processed_variable` to work in the case when `len(t_sol)=1` and also makes it so you no longer need to pass `t` to the call in the cases where the variable is constant (doesn't depend on `t` or `y`) or when `len(solution.t)=1` (e.g. algebraic models)

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #1006 and #571 

EDIT: Note that for variables discretised using finite elements (in y and z) the result returned b [... cut]
closed 2020-05-27 by rtimms