pybamm-team/PyBaMM #1012: CI tests are not run on PRs from forks

tinosulzer, 2020-05-21:
When a PR is made from a fork, the CI tests on GitHub actions do not start. See #1009 and #1010 for example. I don't know if this is due to a temporary outage or a bug in the workflow

dalonsoa, 2020-05-21:
Sorry! I dind't realised this was assigned to @tlestang and have open a PR with - I think - the solution.

tlestang, 2020-05-21:
No worries! I'm just finding out about this issue :)

tlestang, 2020-05-21:
I guess CI was only triggered by pushes?

dalonsoa, 2020-05-21:
Yes, that was the issue. I've just added `pull_request` to trigger it also when there's a PR. In principle, it will trigger then theres a PR against ANY branch, but I guess in practive it will be just develop and, from time to time, master.

PR #1013 (2020-05-21): Trigger GitHub Actions on PR
# Description

Enable triggering of GitHub Actions on PR.

Fixes #1012

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ fl [... cut]
closed 2020-05-21 by tinosulzer