pybamm-team/PyBaMM #1029: Cleaning up geometries and meshes

tinosulzer, 2020-05-28:
The geometry and meshes are quite messy and hard to understand right now. Some ideas for cleaning them up:

- Get rid of lists of submeshes. I think this originally had two purposes:
1. Allow varying submeshes (e.g. 10 r-grid points at x=0, 20 r-grid points at x=0.1, etc). I don't think we need to support this behaviour going forward. We can just say that we only support nice regular geometries. And a lot of the spatial methods just assume that everything behaves like submesh[0] anyway.
2. Record how many submeshes there are. We should be able to read this from `auxiliary_domains` instead.
3. Any other reasons @Scottmar93 @rtimms ?
- Clean up geometries to get rid of "primary", "secondary", etc. Again, if we use `auxiliary_domains` more, we don't need to distinguish between the geometries so much. The only distinction required will be 0-1-2D current collectors. So the messy `Geometry` class can just be replaced by a single function
```python3
def battery_geometry(current_collector_dimension):
```

This will be challenging to implement but should be doable. But it's possible that I'm missing something and this will be fundamentally too difficult.

rtimms, 2020-05-29:
Think this sounds great and I can't think of any reasons not to do it off the top of my head. I think the only time the submesh list is used is to get its length and as you say we should be able to get this from auxiliary domains.

The keys "primary" etc. are used in `meshes` to calculate the total number of points and so on. Will the new framework include auxiliary domains as part of the geometry (at the moment auxiliary domains is a feature of a symbol)? We'll still need some way of getting the right number of repeats. I guess this is what you are getting at in your second bullet

tinosulzer, 2020-05-29:
Yeah, the discretisation would have to read `self.mesh[auxiliary_domains["secondary"][0]]` to find the number of repeats. But it's starting to have to do this anyway, for example for #632 we end up with some variables where primary domain is "negative particle" and secondary domain is "current collector", but the number of mesh repeats assumes that secondary domain is "negative electrode", so you get size errors.

PR #1032 (2020-06-01): Issue 1029 reformat geometry
# Description

Reformats and simplifies geometries. See discussion in #1029 

The geometry no longer has keys "primary" and "secondary". The first keys can either be spatial variables or "tabs", e.g.
```python3
geometry = {
        "negative electrode": {var.x_n: {"min": 0, "max": l_n}},
        "separator": {var.x_s: {"min": l_n, "max": l_n + l_s}},
        "positive electrode": {var.x_p: {"min": l_n + l_s, "max": 1}},
                "negative particle": {var.r_n: {"min": 0, "max": 1}},
                "positive particle": {var.r_p: {"min": 0, "max": 1}},
        [... cut]
closed 2020-06-01 by tinosulzer