pybamm-team/PyBaMM #1033: add list to parameter sets docs

rtimms, 2020-05-31:
Add a list of the available AuthorYear parameter sets in the docs for parameter_sets. At the moment there is no obvious way of seeing what sets are available other than opening the parameter_sets file. Is there a sensible way to automate this?

tinosulzer, 2020-05-31:
Yeah this should be automated so it doesn't get out of date .It would probably be pretty easy to automate once you figure out how to make the github workflow run a script and push the result back. One for @tlestang ?

tlestang, 2020-06-01:
Sounds good, I'll think about it

priyanshuone6, 2021-02-03:
Could you please elaborate more on this?

rtimms, 2021-02-03:
We currently have a number of parameters sets which are available by calling `pyabmm.parameter_sets.AuthorYear` (see [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/parameter_sets.py).  Details about these sets have been added manually to the documentation, but it would be good to automate this so that you don't have to manually update the docstring every time a new parameter set is added.

priyanshuone6, 2021-02-03:
Can I work on this issue because @tlestang is already assigned to it?

tinosulzer, 2021-02-03:
I don't think he is actively working on this right now

priyanshuone6, 2021-02-04:
Please confirm, if I am getting it right

If AuthorYear like this is added in the `parameter_sets.py`
https://github.com/pybamm-team/PyBaMM/blob/8ed813e8406bc2b272a6f2ddc5a4144ff59e6700/pybamm/parameters/parameter_sets.py#L68-L78
then I have to convert these new AuthorYear into this
https://github.com/pybamm-team/PyBaMM/blob/8ed813e8406bc2b272a6f2ddc5a4144ff59e6700/pybamm/parameters/parameter_sets.py#L29-L33 by automating them using github workflow in which a script will run and push the result back in form of a pull request

tinosulzer, 2021-02-04:
Yes, and you can use something like this
https://github.com/pybamm-team/PyBaMM/blob/8ed813e8406bc2b272a6f2ddc5a4144ff59e6700/pybamm/citations.py#L90-L94
to do the formatting of the citation

priyanshuone6, 2021-02-07:
In which folder should I place the script.py?

PR #1371 (2021-02-11): add script and workflow to update parameter_sets.py
# Description

Add a script to update docstring of `parameter_sets.py`
Add a workflow to automatically run the script and make a pull request when modified 

Fixes #1033 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [x] Optimization (back-end change that s [... cut]
closed 2021-02-11 by tinosulzer