pybamm-team/PyBaMM #1044: Link notebooks to Google Colab

ferranbrosa, 2020-06-09:
**Summary**
I have been looking into Google Colab (https://colab.research.google.com/) and it is pretty easy to run any of the notebooks on GitHub. However, you still need to search for the repository and the given notebook so it is not immediate. I see to options here, let me know what you prefer and if you know others please share them below. 

1. Access the GitHub repository from Colab (https://colab.research.google.com/github/pybamm-team/PyBaMM/blob/master). Unfortunately, it seems that this way looses the subdirectory structure and all the notebooks appeared listed alphabetically.
2. Add Google Colab badges to each notebook (see example [here]( https://colab.research.google.com/github/googlecolab/colabtools/blob/master/notebooks/colab-github-demo.ipynb#scrollTo=8QAWNjizy_3O)). Not sure if that can be automated or we will have to do it manually.
3. Collate a series of introductory notebooks (we already have some) and list them in the Getting Started section of the website, with links to launch them directly in Colab.
4. We can also host the notebooks in Drive directly (that means cloning the repository and having to maintain two copies).

These options are not independent so let me know which options you prefer.

ferranbrosa, 2020-06-09:
Personally, I think we should include a link or a badge in the README file in `examples/notebooks` (next to the binder one) to launch option 1. I also vote to implement option 3 on the website.

tinosulzer, 2020-06-10:
That sounds like the best option to me too

ferranbrosa, 2020-06-10:
I have now noticed that if we open a notebook in Colab we need to install PyBaMM every single time (unless it is installed locally by the user, which I think it can be done). I am not sure there is an easy way around this for option 1 as we either modify all the notebooks to include `%pip install pybamm` just before `import pybamm` or we tell users to add this line at the beginning if they use Colab. I am not fully convinced by any of the solutions. The first one is not very elegant if people run things locally (even though in that case all the requirements should be met and the process is really quick) while the second one requires an extra effort from the user which might confuse new users.

ferranbrosa, 2020-06-12:
I don't think this will be too slow if PyBaMM is already installed. It produces quite a lot of text though so the notebook looks a bit messy. Is there a way to print only the warnings/errors when pip installing? Would this be a good idea?

tinosulzer, 2020-06-12:
Can you suppress it with a semicolon?

tinosulzer, 2020-06-12:
Looks like `pip install pybamm -q` should work

ferranbrosa, 2020-06-15:
I have addressed this, adding the extra installation `!pip install pybamm -q` before `import pybamm`. It runs fine both in Jupyter notebook (in my laptop) and Google Colab, but when I push, the tests throw the following error:
```
Traceback (most recent call last):
  File "<string>", line 21, in <module>
NameError: name 'get_ipython' is not defined
```
Is that a problem in the notebooks or is it something we need to fix in the actions?

@dalonsoa @tlestang

tinosulzer, 2020-06-16:
If you look in the tests the line says `get_ipython().system('pip install pybamm -q    # install PyBaMM if it is not installed')` (has it been automatically converted?). You could try importing `get_ipython` but based on [this discussion](https://stackoverflow.com/questions/32538758/nameerror-name-get-ipython-is-not-defined) that might not work

PR #1061 (2020-06-21): Issue 1044 google colab
# Description

Added an extra line to notebooks (before importing PyBaMM) to install PyBaMM (if not installed) so they can be run in Google Colab. I fixed `run-tests.py` to catch if a line in a notebook had magic so ipython is called (before it only caught if a whole cell had magic).

Fixes #1044 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] N [... cut]
closed 2020-06-21 by tinosulzer