pybamm-team/PyBaMM #1065: error message inquiry

gyouhoc, 2020-06-21:
I have following error and warning messages. 
Would you provide me some description? 

Also, I am testing the model with different Maximum concentration in negative and positive electrode and initial concentration in positive and negative electrodes. When I have these errors and warnings what is your advice to mitigate them? 

1. 

At t = 0.20091, , mxstep steps taken before reaching tout.
At t = 0.20091, , mxstep steps taken before reaching tout.
At t = 0.20091, , mxstep steps taken before reaching tout.
At t = 0.20091, , mxstep steps taken before reaching tout.

2. 

SolverError: 
 Maximum number of decreased steps occurred at t=0.1997779210565148. 
Try solving the model up to this time only or reducing dt_max.

tinosulzer, 2020-06-21:
@rtimms ?

tinosulzer, 2020-06-21:
renamed in case anyone else has this issue

rtimms, 2020-06-22:
Hi @gyouhoc , you are probably getting these errors as the solver is stepping over/getting close to an event where something is getting very small or large and causing problems (e.g. concentration going negative). There are a few things you could try 

1) you could integrate up to the time when you get the error and the plot/investigate the solution to see what is happening. Note the time in the error is dimensionless, so you'll need to get the model timescale, e.g.
```
import pybamm
import numpy as np

model = pybamm.lithium_ion.DFN()

# get timescale
param = model.default_parameter_values
timescale = param.evaluate(model.timescale)

# try solving up to the time the solver failed
t_end = 0.19*timescale
t_eval = np.linspace(0, t_end, 100)

sim = pybamm.Simulation(model)
sim.solve(t_eval=t_eval) 
```

2) you could reduce `dt_max`, which is the maximum step size (in seconds) the solver is allowed to take in "safe" mode, e.g.
```
import pybamm

model = pybamm.lithium_ion.DFN()

# get timescale
param = model.default_parameter_values
timescale = param.evaluate(model.timescale)

# work out defualt timescale for your params  
dt_max_default = 0.01 * timescale  # works out as 225 seconds
# try reducing dt_max
solver = pybamm.CasadiSolver(dt_max=60)

sim = pybamm.Simulation(model, solver=solver)
sim.solve()
```

3) you could increase the number of steps the solver is allowed to take before giving up by passing an option to the solver, e.g.
```
solver = pybamm.CasadiSolver(extra_options_setup={"max_num_steps": 10000})
```

4) you could try another solver, or try using "old safe" mode in casadi which works a little differently and may be able to handle this case, e.g.
```
solver = pybamm.CasadiSolver(mode="old safe")
```

Let us know how you get on, as there may be something we can change in the solver that helps avoid this, or at least makes it more obvious what to try when you see these errors.

rtimms, 2020-06-22:
I should add that `dt_max` is more like a global maximum timestep. The casadi solver doesn’t support events of of the box, so the way event handling has been implemented is that it uses adaptive timestepping to go from `t` to `t+dt_max` and then checks to see if events have been triggered. If they have it should go back and halve `dt_max` and try to integrate again to hone in on the event. What’s probably happening is that the solver breaks in some non-recoverable way when trying to go from `t` to `t+dt_max`. Hopefully reducing dt_max will help

tinosulzer, 2020-06-24:
We should find a way to catch this error and suggest increasing `dt_max` while printing the current value of `dt_max`.

felipe-salinas, 2020-07-15:
One way I have been working my way out from that problem, in a case like the one @rtimms mentions (electrolyte concentration approaching zero/negative values), has been increasing the points in the grid (var_pts) and decreasing the solver tolerance (rtol, atol), using the commands in:
`sim = pb.Simulation(var_pts=var_pts)`
`sim.solve(solver=pb.CasadiSolver(mode="safe", rtol=1e-9, atol=1e-9))`

closed 2020-09-18 by gyouhoc

Tangzl922, 2022-07-20:
Hi,
I also met this problem and I tried all the aforementioned methods but it still doesn't work. Is there already a general solution for this problem?

Regards,

psk1998, 2023-07-13:
> One way I have been working my way out from that problem, in a case like the one @rtimms mentions (electrolyte concentration approaching zero/negative values), has been increasing the points in the [...]

Thanks, this worked for me.