pybamm-team/PyBaMM #1068: Discarge capacity does not decrease with C rate

tanhweejien, 2020-06-23:
I expected the discharge capacity to decrease with increasing C-rate, because the electrolyte and electrodes have finite resistance. What I see in the result is that the discharge capacity stayed relatively constant even up to 8C. May I know where in the code did I do wrong? 

```
import pybamm
ecker_set=pybamm.parameter_sets.Ecker2015
params_ecker=pybamm.ParameterValues(chemistry=ecker_set)
model=pybamm.lithium_ion.DFN(name='DFN_Ecker')
sim=pybamm.Simulation(model=model, parameter_values=params_ecker)
params=sim.parameter_values
```

`for factor in range(1,9):`
    
    params.update({
    'Current function [A]':0.17101*factor,
    'Cell capacity [A.h]':0.17101
    })
    sim.specs(parameter_values=params)
    sim.solve()
    print('*'*20)
    print('End time: {} h'.format(sim.solution['Time [h]'].entries[-1]))
    print('Termination: {}'.format(sim.solution.termination))
    print('Factor: '+ str(factor))
    print('Current function: {} A'.format(sim.parameter_values['Current function [A]']))
    print('Discharge capacity: {} A.h'.format(sim.solution['Discharge capacity [A.h]'].entries[-1]))
    print('V_terminal_at_EOD: {} V'.format(sim.solution['Terminal voltage [V]'].entries[-1]))
    print('Cut-off voltage: {} V'.format(sim.parameter_values['Lower voltage cut-off [V]']))`

Here is the output:
********************
End time: 1.0 h
Termination: final time
Factor: 1
Current function: 0.17101 A
Discharge capacity: 0.17100999999999988 A.h
V_terminal_at_EOD: 2.5193501737862185 V
Cut-off voltage: 2.5 V
********************
End time: 0.5 h
Termination: final time
Factor: 2
Current function: 0.34202 A
Discharge capacity: 0.1710100000000006 A.h
V_terminal_at_EOD: 3.086274150325055 V
Cut-off voltage: 2.5 V
********************
End time: 0.3333333333333333 h
Termination: final time
Factor: 3
Current function: 0.51303 A
Discharge capacity: 0.17101000000000016 A.h
V_terminal_at_EOD: 3.076588567832502 V
Cut-off voltage: 2.5 V
********************
End time: 0.25 h
Termination: final time
Factor: 4
Current function: 0.68404 A
Discharge capacity: 0.1710100000000001 A.h
V_terminal_at_EOD: 3.016583755405115 V
Cut-off voltage: 2.5 V
********************
End time: 0.2 h
Termination: final time
Factor: 5
Current function: 0.85505 A
Discharge capacity: 0.1710099999999995 A.h
V_terminal_at_EOD: 2.952188564587493 V
Cut-off voltage: 2.5 V
********************
End time: 0.16666666666666666 h
Termination: final time
Factor: 6
Current function: 1.02606 A
Discharge capacity: 0.17101 A.h
V_terminal_at_EOD: 2.8740390964996774 V
Cut-off voltage: 2.5 V
********************
End time: 0.14285714285714285 h
Termination: final time
Factor: 7
Current function: 1.19707 A
Discharge capacity: 0.1710099999999997 A.h
V_terminal_at_EOD: 2.81435300000385 V
Cut-off voltage: 2.5 V
********************
End time: 0.125 h
Termination: final time
Factor: 8
Current function: 1.36808 A
Discharge capacity: 0.17101000000000016 A.h
V_terminal_at_EOD: 2.7219156569932172 V
Cut-off voltage: 2.5 V

tanhweejien, 2020-06-23:
I have also computed something similar using DandeLiion for comparison. Please see below:

[image]

tinosulzer, 2020-06-23:
Weird. Running the `rate_capability.py` example with the Ecker parameters I get these plots.
[image]
[image]

ferranbrosa, 2020-06-23:
Given that you hit final time rather than cut-off voltage, may it be that the default running time underestimates discharge time and therefore you always pull out the same amount of charge?

What happens if you define an experiment that says the following?
```
"Discharge at {}C until 2.5 V (5 seconds period)".format(Crate),
```

Note that you might need to play around with the period time to make sure you get close to the critical voltage. You can check how it is implemented in the `rate_capability.py` example [here](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/scripts/rate_capability.py) as Tino suggested.

rtimms, 2020-06-23:
yes @ferranbrosa I think either something like that is happening, or the way the parameters are being set isn't quite right. I haven't tried the script above, but running the below I get the following discharge curves (note you need to play with the number of points/range in t_eval in order to have enough resolution to pick up the tails at the end. you could probably do with even more points than I have here to pick up the sharp drop in the end in the outputted solution data)

```
import pybamm
import numpy as np

pybamm.set_logging_level("INFO")
model = pybamm.lithium_ion.DFN()

chemistry = pybamm.parameter_sets.Ecker2015
parameter_values = pybamm.ParameterValues(chemistry=chemistry)
current_1C = parameter_values["Cell capacity [A.h]"]
currents = np.array([0.1, 1, 2, 5, 10]) * current_1C
parameter_values["Current function [A]"] = "[input]"

solver = pybamm.CasadiSolver(dt_max=1)
t_eval = np.linspace(0, 4000, 100)
sim = pybamm.Simulation(model, parameter_values=parameter_values)

sols = [None] * len(currents)
for i, current in enumerate(currents):
    sols[i] = sim.solve(t_eval=t_eval, inputs={"Current function [A]": current})

pybamm.dynamic_plot(sols, ["Terminal voltage [V]"], time_unit="seconds")
```

[Figure_1.pdf](https://github.com/pybamm-team/PyBaMM/files/4820578/Figure_1.pdf)

tanhweejien, 2020-06-24:
Thanks for all the input!!

I just realised a mistake I made. When I used DandeLiion, the C rates were defined wrt. the time needed to discharge, ie. 1C referred to the current to discharge the cell to cut off voltage in 1h, 2C referred to the current to discharge the cell to cut off voltage in 1/2h etc. Due to the finite resistance of the electrolyte and the activation overpotential, the relationship between C rate and current is not linear. When I simulated the cell using PyBaMM, I somehow assumed linear relationship between C rate and current (thus the multiplying factor in my code). That said, something still does not feel quite right. I will look into this. Can we please keep this open for the time being?

tanhweejien, 2020-06-25:
I have a question regarding t_eval.
I am assuming that t_eval default value is 1h.
If that is the case, it is weird that the termination for discharge at C rate higher than 1C is 'final time' rather than 'lower cut off voltage'. If we discharge at C rate higher than 1C, we might reach lower cut off voltage even before we hit the 1h limit.
I think I dont understand t_eval properly, can anyone please explain?

tinosulzer, 2020-06-25:
Yeah there is definitely something going wrong there. I can't reproduce this. Can you pull the latest changes, and create a new simulation each time instead of using `sim.specs`? We are about to deprecate `sim.specs`

tinosulzer, 2020-06-25:
Ok, so your example has highlighted some bugs we need to fix:
1. Using the same model in several simulations can sometimes mean that the parameters don't get updated properly (to be fixed in #1011 
2. The casadi solver doesn't handle events very precisely (it doesn't include the final state where the event occurs in the solution, to be fixed in #1082 )
3. The casadi solver doesn't do well with interpolation (e.g. interpolating OCPs like in the Ecker set). Less of a bug as there isn't much we can do about it, more of an observation

In the meantime, the following script will work better (using input parameters and the IDA KLU solver)
```python3
import pybamm
import numpy as np

# pybamm.set_logging_level("DEBUG")
ecker_set = pybamm.parameter_sets.Ecker2015
params_ecker = pybamm.ParameterValues(chemistry=ecker_set)
model = pybamm.lithium_ion.DFN(name="DFN_Ecker")

# set mesh
var = pybamm.standard_spatial_vars
var_pts = {var.x_n: 10, var.x_s: 10, var.x_p: 10, var.r_n: 30, var.r_p: 30}
params_ecker.update(
    {
        "Current function [A]": 0.17101 * pybamm.InputParameter("factor"),
        "Cell capacity [A.h]": 0.17101,
    }
)

sim = pybamm.Simulation(
    model=model,
    parameter_values=params_ecker,
    var_pts=var_pts,
    solver=pybamm.IDAKLUSolver(),
)

for factor in range(1, 9):
    sim.solve(t_eval=np.linspace(0, 3600 / factor, 10), inputs={"factor": factor})
    print("*" * 20)
    print("End time: {} h".format(sim.solution["Time [h]"].entries[-1]))
    print("Termination: {}".format(sim.solution.termination))
    print("Factor: " + str(factor))
    print("Current function: {} A".format(sim.parameter_values["Current function [A]"]))
    print(
        "Discharge capacity: {} A.h".format(
            sim.solution["Discharge capacity [A.h]"].entries[-1]
        )
    )
    print("Current: {} A".format(sim.solution["Current [A]"].entries[-1]))
    print(
        "V_terminal_at_EOD: {} V".format(
            sim.solution["Terminal voltage [V]"].entries[-1]
        )
    )
    print(
        "Cut-off voltage: {} V".format(
            sim.parameter_values["Lower voltage cut-off [V]"]
        )
    )
```
For me, it gives this output:
```
********************
End time: 0.9648313073905167 h
Termination: event: Minimum voltage
Factor: 1
Current function: 0.17101 * factor A
Discharge capacity: 0.16499580187685237 A.h
Current: 0.17101 A
V_terminal_at_EOD: 2.4999999999999996 V
Cut-off voltage: 2.5 V
********************
End time: 0.4843824852013818 h
Termination: event: Minimum voltage
Factor: 2
Current function: 0.17101 * factor A
Discharge capacity: 0.16566849758857616 A.h
Current: 0.34202 A
V_terminal_at_EOD: 2.4999999999999982 V
Cut-off voltage: 2.5 V
********************
End time: 0.32282910506617973 h
Termination: event: Minimum voltage
Factor: 3
Current function: 0.17101 * factor A
Discharge capacity: 0.1656210157721027 A.h
Current: 0.51303 A
V_terminal_at_EOD: 2.49999999999969 V
Cut-off voltage: 2.5 V
********************
End time: 0.23856098684879842 h
Termination: event: Minimum voltage
Factor: 4
Current function: 0.17101 * factor A
Discharge capacity: 0.16318525744405168 A.h
Current: 0.68404 A
V_terminal_at_EOD: 2.4999999999998295 V
Cut-off voltage: 2.5 V
********************
End time: 0.18601868707265934 h
Termination: event: Minimum voltage
Factor: 5
Current function: 0.17101 * factor A
Discharge capacity: 0.1590552783814773 A.h
Current: 0.85505 A
V_terminal_at_EOD: 2.499999999999999 V
Cut-off voltage: 2.5 V
********************
End time: 0.15030442551768866 h
Termination: event: Minimum voltage
Factor: 6
Current function: 0.17101 * factor A
Discharge capacity: 0.15422135884667929 A.h
Current: 1.02606 A
V_terminal_at_EOD: 2.499999999999888 V
Cut-off voltage: 2.5 V
********************
End time: 0.12452883057212981 h
Termination: event: Minimum voltage
Factor: 7
Current function: 0.17101 * factor A
Discharge capacity: 0.14906972721297934 A.h
Current: 1.19707 A
V_terminal_at_EOD: 2.4999999999999996 V
Cut-off voltage: 2.5 V
********************
End time: 0.105044151278123 h
Termination: event: Minimum voltage
Factor: 8
Current function: 0.17101 * factor A
Discharge capacity: 0.14370880248057474 A.h
Current: 1.36808 A
V_terminal_at_EOD: 2.4999999999999174 V
Cut-off voltage: 2.5 V
```
Let me know if this is satisfactory and we can close this issue as the bugs are already documented in other issues

tanhweejien, 2020-06-25:
Is the IDA KLU solver only available to Linux/Mac OS? (I am using window)
Other than that, I think this issue can be closed. Thank you so much!!

tinosulzer, 2020-06-26:
I don't know about getting the IDA KLU solver directly on windows (@tlestang ?) but you can definitely install it if you install through [WSL](https://pybamm.readthedocs.io/en/latest/install/windows.html)

closed 2020-06-26 by tinosulzer