pybamm-team/PyBaMM #1069: Calculating Jacobian failed + the corrector convergence failed repeatedly or with |h| = hmin

tanhweejien, 2020-06-23:
I am getting the errors shown in the title when I am simulating discharge at high C rate (10C). I still get solution but I dont know if this means the solution is also erroneous?

tinosulzer, 2020-06-23:
Can you post your code? If you do "```python3 " before the code block it will format nicely

tanhweejien, 2020-06-23:
10C

```python3
import pybamm
ecker_set=pybamm.parameter_sets.Ecker2015
params_ecker=pybamm.ParameterValues(chemistry=ecker_set)
model=pybamm.lithium_ion.DFN(name='DFN_Ecker')
sim=pybamm.Simulation(model=model, parameter_values=params_ecker)
params=sim.parameter_values
params.update({
    'Current function [A]':0.17101*10,
    'Cell capacity [A.h]':0.17101
    })
sim.specs(parameter_values=params)
sim.solve()
```
11C

```python3
import pybamm
ecker_set=pybamm.parameter_sets.Ecker2015
params_ecker=pybamm.ParameterValues(chemistry=ecker_set)
model=pybamm.lithium_ion.DFN(name='DFN_Ecker')
sim=pybamm.Simulation(model=model, parameter_values=params_ecker)
params=sim.parameter_values
params.update({
    'Current function [A]':0.17101*11,
    'Cell capacity [A.h]':0.17101
    })
sim.specs(parameter_values=params)
sim.solve()
```

tinosulzer, 2020-06-23:
Feel free to disregard this. What is happening is that, the way events are implemented, the casadi solver sometimes steps into infeasible regions (e.g. c_s < 0) before the events are triggered. If you want to convince yourself this is fine or get rid of the error, you can reduce the time `t_eval` (in seconds) for which the model is being solved (default is 1 hour)
```python3
import pybamm
import numpy as np

ecker_set = pybamm.parameter_sets.Ecker2015
params_ecker = pybamm.ParameterValues(chemistry=ecker_set)
model = pybamm.lithium_ion.DFN(name="DFN_Ecker")
sim = pybamm.Simulation(
    model=model, parameter_values=params_ecker  # , solver=pybamm.IDAKLUSolver()
)
params = sim.parameter_values
Crate = 10
params.update({"Current function [A]": 0.17101 * Crate, "Cell capacity [A.h]": 0.17101})
sim.specs(parameter_values=params)
sim.solve(t_eval=np.linspace(0, 3600 / Crate))
sim.plot()
```
I'll leave this issue open until we find a way to hide this warning

tinosulzer, 2020-11-04:
After further consideration, I think it is a bad idea to hide this warning, since it is possible (and recommended) to change either the integration timespan or the solver settings (`dt_max`) to avoid the warning. See [this new notebook](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/speed-up-solver.ipynb) for details.
Closing the issue now

closed 2020-11-04 by tinosulzer