pybamm-team/PyBaMM #1076: Error:""Use an equivalent CasADi function instead of that numpy function."

zhenghao87, 2020-06-24:
**Describe the bug**
When running the example code:

"
```
import pybamm
model = pybamm.lithium_ion.DFN() # Doyle-Fuller-Newman model
sim = pybamm.Simulation(model)
sim.solve()
```
"
produce following error:

"
```
~\AppData\Local\Continuum\anaconda3\envs\pybamm-env36\lib\site-packages\casadi\casadi.py in __array__(self, *args, **kwargs)
  10826             raise Exception("Implicit conversion of symbolic CasADi type to numeric matrix not supported.\n"
  10827                        + "This may occur when you pass a CasADi object to a numpy function.\n"
> 10828                        + "Use an equivalent CasADi function instead of that numpy function.")
  10829 
  10830 

Exception: Implicit conversion of symbolic CasADi type to numeric matrix not supported.
This may occur when you pass a CasADi object to a numpy function.
Use an equivalent CasADi function instead of that numpy function.
```
"

Systeminfo:

Anaconda 5.2, python3.6,  windows10

Thanks!

[image]

rtimms, 2020-06-24:
I think this was fixed by #1067 . Could you try getting the latest version from the `develop` branch and see if you still get the same error?

tinosulzer, 2020-06-26:
@zhenghao87 did this work?

ferranbrosa, 2020-07-02:
@zhenghao87 the new release (v0.2.3) should fix this now

closed 2020-07-02 by tinosulzer