pybamm-team/PyBaMM #1079: Website Getting Started notebooks

ferranbrosa, 2020-06-24:
**Summary**
Now that Google Colab compatibility is about to go live it would be nice to create an introductory set of notebooks in the website (Getting Started) section which would launch directly in Colab. Might be a good time to think if we want to update/extend the Getting Started notebooks from the experience we got in the workshops.

 **Motivation**
Set up a very user-friendly set of notebooks to lower the entry barrier to PyBaMM.

ferranbrosa, 2020-06-25:
Here is a possible structure for the notebooks:

1. How to run a simulation: basically you define a model, a simulation and run it (the basic stuff, kind of "Hello world!").
2. Compare models: again, very few lines but we are introducing the three standard models.
3. Basic plotting: it also helps introducing the variables so we don't have to do it later.
4. Change parameters: in the same script.
5. Run experiments: quite descriptive.
6. Advanced model features: model options, access variables, store outputs...
7. Advanced solver features: change mesh resolution and solver tolerance.

tlestang, 2020-06-26:
I agree that having some ready-to-try examples on the website would do a lot of good.
Other than notebooks, the new kid in the block is [https://www.streamlit.io/](streamlit.io) which could be useful to showcases some potential use cases of pybamm

ferranbrosa, 2020-06-26:
That looks really cool! Might be useful in the future to write some scripts with some data analysis features. @TomTranter would that be helpful for the GUI?

PR #1083 (2020-06-30): Issue 1079 update getting started notebooks
# Description

I have reformatted the Getting Started notebooks adding some content from the Faraday Tech notebooks.

Fixes #1079 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-t [... cut]
closed 2020-06-30 by tinosulzer