pybamm-team/PyBaMM #1082: Reformat casadi solver again

tinosulzer, 2020-06-25:
Another reformatting of the casadi solver is needed with

1. Better automatic choosing of `dt_max` (`dt_max = 0.1 * min(timescale, t_final)`
2. Generating the integrator once with `[0,1]` and rescaling time
3. Look into a smarter way of dealing with events ("The alternative is to try to be smarter about predicting when events will occur. At constant current you can see most of our events coming from a long way away (i.e. they are pretty much linear, definitely monotonic, in time). So you could envisage an algorithm where after each step you look at the gradient of the event function, look at where the function would cross zero, and choose the size of your next step based on that.")

Anything else?

tinosulzer, 2021-01-18:
From discussion this morning with @rtimms , we should look into whether the casadi solver can give information on when it fails. This would allow us to do away with `dt_max`: first try solving over the whole interval, and if there is a failure then solve up to the point where it fails (and check when/if the event was crossed)

rtimms, 2021-01-18:
Had a little play with this using a solution from [here](https://stackoverflow.com/questions/5136611/capture-stdout-from-a-script) to capture stdout, so seems like something along these lines will work. One thing to be careful of is that you don't fall into the trap of not having enough points in `t_eval` to do good event detection... I guess if an event is crossed between two times `t_{i}` and `t_{i+1}` you could use the "go back and dense solve" method in that interval to get an accurate solution there. As discussed, there is probably a much better way of locating the events.

tinosulzer, 2021-01-20:
If we manage to make this work then we should rethink the default max simulation time in the experiment when doing "until" conditions (currently 1 week, could probably be shorter by default, or based on C-rate)

tinosulzer, 2021-02-15:
Another idea: make the rhs "rhs * switch" where "switch" is a heaviside that turns off when an event is crossed; make the algebraic "algebraic * switch + y * (1- switch)"

tinosulzer, 2021-02-16:
This doesn't work with all events as some have a `min` or a `max`, which aren't differentiable (can't calculate jacobian).
Restricting to "voltage" and "current" events for now

PR #1450 (2021-03-29): Issue 1082 events
# Description

Add a new casadi solver mode which provides a fast solution even when there are events. Eventually this could become the default mode for the casadi solver, but it needs more testing first.

Should not be part of v0.4.0, do not merge before #1448 

Fixes #1082 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking ch [... cut]
closed 2021-03-29 by tinosulzer