pybamm-team/PyBaMM #1085: Dimensional external varibales

rtimms, 2020-06-29:
Most user-facing aspects of pybamm now deal with dimensional quantities (e.g. parameters, `t_eval`), so it would be good to make it so you can pass dimensional variables to the `external_variables` dict when solving. This requires a little bit of converting as it is the dimensionless variables that appear in the state vector, but should be doable

tinosulzer, 2020-06-29:
How do you suggest automating this?

rtimms, 2020-06-29:
I can only think of a very hacky implementation where you look for the string name with/without [dimensions] then get the scales by evaluating the both the dimensional and dimensionless variables at two arbitrary states, then you can solve for the multiplicative and additive scales in the non-dimensionalisation. This really isn't very nice though. One option would be to store details of the non-dim on the variable object (or somwhere else) so that we can have a method that converts between dim/non-dim

TomTranter, 2020-07-01:
It would be handy for the non-dimensionalisation to be more visible to the user. Could you use the units in the variable name to do the non-dim automatically - I don't know if the actual values you pick for certain variables matter that much if they have the same units e.g. pos and neg particle conc. can you use one reference conc. for both?

rtimms, 2020-07-02:
There are things like pos and neg potentials are defined with respect to some reference which is different for pos and neg. I think there are quite a few examples where units alone isn't enough to give you the non-dim we used. I think ideally we would store the scale or a method on the  non-dim variables that converts to dimensional for you

TomTranter, 2020-07-02:
Yeah but did you choose different values for a good "enough" reason numerically. I could see that if the potential gradients varied differently by more than a few orders of magnitude you would non-dim differently but might simplify things not to. I'm thinking if the typical length, typical temperature, typical voltage etc were more accessible i.e. all in the parameters then it would make things more transparent and if you could automate the non-dimensionalisation more you could reduce the code complexity and also the amount of data being stored. Storing the method on the variable is a good idea. Maybe it could use the defaults but have the ability to override certain values

TomTranter, 2020-07-02:
I guess with the concentrations for example you would pick the max conc for each particle and then the non dim would represent the soc in a way that is sort of meaningful which you would lose

rtimms, 2020-07-04:
Yeah I think we should store the scales. It makes sense to scale things differently even if they have the same units. For example, we scale potentials multiplicatively by the thermal voltage, but the negative potential is close to 0V and the positive potential close to some other value (e.g. 3V, depends on chemistry) so it makes sense to measure them with respect to a different reference value.

Storing the scales used by the non-dim vars makes sense and as you say we could then automate conversion between dim and non-dim and reduced the size of the variables dict. Having said that, it is useful to be able to search it and see what variables are available as some quantities are stored in a range of different units e.g.  
```
'Electrolyte concentration',
'Electrolyte concentration [mol.m-3]',
'Electrolyte concentration [Molar]',
```

dalonsoa, 2021-02-04:
Not sure if relevant after so much time, but I have used [Pint](https://pint.readthedocs.io/en/stable/) in the past and it seems to work pretty well to handle magnitudes with units (once you get used to it). Internally probably is of no use, but for the user-facing part, might be handy.

closed 2022-12-05 by tinosulzer