pybamm-team/PyBaMM #1098: Not all parameters can be used as input parameters

TomTranter, 2020-07-02:
**Summary**
Some parameters go into evaluating timescales and maybe other things and are not allowed as input parameters. Electrode height used to be ok but now doesn't work and is quite useful - in fact essential for the jelly roll model. Can we try to put this functionality back in. For this particular one the problem is in base_solver when this line is called
```
model.timescale_eval = model.timescale.evaluate()
```
in the set_up function

set_up is actually passed inputs so could use them to evaluate but there are also other problems further down the line with the plotting function

If we decide to not allow some parameters to be inputs we should at least distinguish between flexible and fixed parameters that require a brand new model a bit more formally.

rtimms, 2020-07-02:
you can work around this by first setting model.timescale using some sensible reference height. we want to discourage changing parameters that change the timescale as this might cause some problems

TomTranter, 2020-07-02:
Once model .timescale_eval is set though is that used exclusively instead of calling model.timescale.evaluate(). I'd be worried hacking it once then calling the function later would also cause issues

rtimms, 2020-07-02:
What I meant was to work out a sensible timescale and then set model.timescale to pybamm.Scalar(value), rather than changing timescale_eval

tinosulzer, 2020-07-02:
That's a good idea! Avoids having to be clever. You could use the value of the timescale before you change the parameters to inputs, by doing `param.evaluate(model.timescale)`. Should we edit the error message to suggest this?

rtimms, 2020-07-02:
yeah definitely update the error message to explain this

TomTranter, 2020-07-03:
The timescale also involves i_typ which is a current density so also uses the height and width. Are we suggesting having a typical value for every input parameter which sounds like it would be more suited to being a feature of the input parameter itself rather than a similarly named separate parameter

rtimms, 2020-07-03:
I think we are suggested that you do something like the following 

```python3
model = pybamm.lithium_ion.SPM()

# evaluate timescale using default params
params = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)
timescale = params.evaluate(model.timescale)
model.timescale = pybamm.Scalar(timescale)

# now set inputs
params["Electrode height [m]"] = "[input]"

sim = pybamm.Simulation(model, parameter_values=params)
sim.solve([0, 3600], inputs={"Electrode height [m]": 0.06})
sim.plot()
```

This solves ok, but plotting fails as it tries to evaluate the scale used for 'y' and 'z'. This would be fixed by only evaluating the scales for the domains used in the model (or only adding spatial variables 'y' and 'z' to 'model.variables' if `dimensionality` is 1 or 3). More generally we should raise errors for any parameters that appear in time or length scales and suggest that the user picks a fixed value for these.

I don't think we need a typical value for all input parameters, we just need to be careful if people are changing parameters used in the non-dimensionalisation on the fly.

rtimms, 2020-07-03:
wait sorry I see what you aren’t getting at now - variables might be getting sales differently when the inputs aren’t different. It shouldn’t matter if you are comparing dimensional things, but the non dim vars are scaled differently. Hmm.. need to think moreno about this

TomTranter, 2020-07-03:
I kinda like the idea of assigning a typical value to each input which is used exclusively for scaling but I don't know how practical that is to implement and manage. It would be far simpler to not allow certain parameters to be inputs for now or just live with the current error messages. I'm not particularly fond of your above hack

TomTranter, 2020-07-03:
But then again it is quite useful to allow height and width as inputs... and it used to work so maybe we could undo the change that broke it

rtimms, 2020-07-03:
I think there is an important distinction here between setting up a simulation once and then solving multiple times with different inputs, in which case it is fine to have things like height as an input, v.s. passing inputs between calling `step` in which case I'm worried things might go wrong if the scalings change... @tinosulzer what are your thoughts? I think you are probably more familiar with how this is implemented.

tinosulzer, 2020-07-03:
To be honest we can probably get away with only setting `model.timescale_eval` if it has not been set yet (using all inputs)

rtimms, 2020-07-04:
Actually, what I suggested won't work at all. There are dimensionless groupings that appear in the governing equations which rely on the fact we explicitly scaled with the discharge timescale, so changing `model.timescale` independently gives the wrong result (see example code at end). As a result I don't think we should allow the timescale to be changed (at least for lithium-ion and lead-acid models where we have made the explicit scaling of time by discharge time in order to get the dimensionless coefficients in the equations).

I think there are 2 use cases for `InputParameter`s:

1. To set up a `Simulation` once and the do separate solves, but with different parameters
2. To change the values of parameters between doing a `step` in the solver

Clearly it doesn't make sense to change things like the cell geometry in case 2, but it might make sense to change other parameters such as the heat transfer coefficient at the boundary. I'm not suggesting we have an "allowed" list of parameters you can change -- I think to some degree it should be up to the user to do things that are physically sensible.

In case 1 it is totally fine to to have `InputParameter`s that appear in the timescale, as you aren't changing the timescale on-the-fly during a simulation -- instead you are creating separate solutions for different parameter values.

I'm not sure what the best solution to this problem is. Maybe we evaluate `model.timescale` once, as @tinosulzer  suggests, and then check that any inputs that appear in the timescale don't change their value between calls to `step` and raise an error if they do. 

@TomTranter could you remind me of how you were using the height as an input? I think as case 1, since each different height represented a different single cell, but then you were stepping within that. Maybe that is a awkward case to check, but I think what you were doing is ok

tinosulzer, 2020-07-07:
I wonder if we could just multiply (divide?) the rhs by `model.timescale` and then solve with the dimensional time. But this probably just shifts the problem to a different point, since the "timescale" will change if things like electrode height are changed.

rtimms, 2020-07-08:
yeah I think this just moves the problem. I think its safest to not let the value of input parameters that appear in the timescale to change within the same solution

TomTranter, 2020-10-21:
So this has not been addressed yet and I've just come across it again trying to update my code to work with latest pybamm branch. In the case of electrode height Rob you are right that I'm using it as an input because I have many models being solved for different sections of the jellyroll and want to re-use the simulation object but during a solve and step it doesn't change for each individual model. Whereas something like temperature is also an input but does change throughout the simulation. In the case of height the timescale should then remain fixed and it shouldn't be a problem so could a solution be that instead of throwing an error we just throw a warning about the timescale containing an input. Or should we be more careful and make a distinction between inputs that change and those that don't.... I guess I'm kinda using the electrode height input in a way that doesn't make a lot of sense calling them inputs. It's more like a parameter but it's useful to have them as variable parameters. I'd really like to avoid having to set-up a simulation for each mini-model

rtimms, 2020-10-21:
So I think a solution to this is to allow all parameters to be inputs and calculate time and length scales in `base_solver`, storing `timescale_eval` and `length_scales_eval` on the solution itself. These can then be used in `ProcessedVariable`. The workflow for this would be:

1. create a model with the inputs you want e.g. electrode height
2. create separate solvers for each electrode height you want to use -- in the init the evaluated time/length scales will be stored as attributes of the solver
3. call the solvers with inputs corresponding to the correct input height. if stepping we check to make sure that the scales havent changed 

now you have a single model, but separate solutions for each electrode height. I guess this means you calculate the jacobian etc multiple times -- @TomTranter not sure if you avoid this in your current implementation?

TomTranter, 2020-10-21:
@rtimms sounds good. I can have a go at implementing this and benchmark for speed. Not sure about calculating jacobian multiple times.

TomTranter, 2020-10-22:
As processed_variable is not passed a solver would it make more sense to store timescale on the solution

rtimms, 2020-10-22:
yeah, that makes sense

tinosulzer, 2020-11-03:
Can we close this?

TomTranter, 2020-11-03:
Yep

closed 2020-11-03 by TomTranter