pybamm-team/PyBaMM #1101: Error with too many children when processing parameter with tabular data

dalonsoa, 2020-07-10:
**Describe the bug**
When updating the positive or negative electrode OCP entropic change with tabular data like in:

```python
params.update({"Negative electrode OCP entropic change [V.K-1]": "[data]my_data"}, path="")
```
There is an error when creating the interpolant in [here](https://github.com/pybamm-team/PyBaMM/blob/b2488ce5cfec4a445e4e42f77530eff5545757cd/pybamm/parameters/parameter_values.py#L511). The traceback is enourmous, but the last bit is:

```
~/Documents/Projects/PyBaMM-GEM/PyBaMM/pybamm/parameters/parameter_values.py in _process_symbol(self, symbol)
    512                     print(i, v)
    513                 name, data = function_name
--> 514                 function = pybamm.Interpolant(data, *new_children, name=name)
    515             elif isinstance(function_name, numbers.Number):
    516                 # If the "function" is provided is actually a scalar, return a Scalar

TypeError: __init__() got multiple values for argument 'name'
```
I've identified that, apparently, when procesing these parameters, `new_children` has actually two elements, but `pybamm.Interpolate` only expect one child. The second child is assigned to the argument `name` and therefore the actual keyword argument `name` is redundant, hence the error. 

**To Reproduce**
Run the following script. Attached is the input file. 
[my_data.csv.zip](https://github.com/pybamm-team/PyBaMM/files/4902407/my_data.csv.zip)

The error is the same when using the default chemistry or the SPMe and DFN models.

```python
import pybamm

exp = pybamm.Experiment(["Discharge at 1C until 2.5V", 
                         "Rest for 1 hour", 
                         "Charge at 1C until 4.2V",
                         "Hold at 4.2V until 50mA",])

model = pybamm.lithium_ion.SPM(options={"thermal": "x-full"}  )

params = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)
params.update({"Negative current collector surface heat transfer coefficient [W.m-2.K-1]": 0, 
               "Positive current collector surface heat transfer coefficient [W.m-2.K-1]": 0}, 
              check_already_exists=False)
params.update({"Negative electrode OCP entropic change [V.K-1]": "[data]my_data"}, path="")

sim = pybamm.Simulation(model=model, parameter_values=params, experiment=exp)
sim.solve()
```

**Expected behaviour**
The parameter should be processed without error. 

**Screenshots**
None

**Additional context**

The two children that appear to be causing the problem are:

```output
broadcast(boundary value(X-averaged positive particle concentration))
Maximum concentration in positive electrode [mol.m-3]
```

tinosulzer, 2020-07-10:
The problem is that when pybamm sees a `[data]` tag it creates an `Interpolant` object, which can usually act as drop-in replacement for a function. But here the function (OCP entropic change) expects two children, the stoichiometry and the maximum concentration), so the interpolant doesn't know what to do. The fix is to create your own function that calls the interpolant, e.g.
```python3
def entropic_change(sto, c_n_max):
    # in this case we ignore c_n_max
    interp = interp1d(sto_data, entropic_data)
    return pybamm.Function(interp, sto)
```
I haven't actually tested this but pretty sure it would work.
If you wanted a 2D interpolant (for example for functions that depend on concentration and temperature) you would have to define that in a custom way also, e.g.
```python3
def fun(c,T):
    interp = interp2d(c_data, T_data, fun_data)
    return pybamm.Function(interp, c, T)
```
We should raise a custom error message explaining this and referencing this issue.

dalonsoa, 2020-07-14:
Many thanks for pointing this. So, just to be clear, in your example above, the function `entropic_change` should be added to its own `entropic_change.py` file and the parameter defined as:

```python
params.update({"Negative electrode OCP entropic change [V.K-1]": "[function]entropic_change"}, path="")
```
right?

I'll try to open a PR with the custom error message as soon as possible. In the meantime, I suggest to to close this issue, even though the solution is now clear.

tinosulzer, 2020-07-14:
You can either do that or create the function in your main file and reference it directly (pass the function object). If you do the first option, you need to set the path to point to where the function is defined.
Did you mean do close or don't close?

dalonsoa, 2020-07-15:
To NOT close this issue. I can't spell 🤦

dalonsoa, 2020-07-17:
I've been trying to implement this and I've come with a a few issues.

### Attempt 1:
If I use your approach directly, using `interp1d`, I get:

```
AttributeError: 'interp1d' object has no attribute '__name__'
```
Being `interp1d` an instance of a class, called via its `__call__` method,  rather than an actual function, it lacks of of the `__name__` attribute that PyBaMM requires. I would assume the same applies to `interp2d` or, in general, any of `scipy` interpolants. 

### Attempt 2:
I've tried to wrap the call to `interp1d` with an actual function:
```python
    interp = interp1d(xdata, ydata)
    def finterp(sto):
        return interp(sto)
    return pybamm.Function(finterp, sto)
```
After a long traceback, I get:
```
Exception: Implicit conversion of symbolic CasADi type to numeric matrix not supported.
This may occur when you pass a CasADi object to a numpy function.
Use an equivalent CasADi function instead of that numpy function.
```
So, this seems like a dead end. Do you have any suggestion to bypass this - keeping aside trying to use other solver or using CasADi lower level functions to do the inteprolation?

tinosulzer, 2020-07-17:
I think both are resolvable:
1. Only needs the name for the purpose of giving a name to the object. There is already a try-catch in `Function.__init__` to use `__class__` if `__name__` can't be found so I'm not sure where that error is coming from
2. You could add `interp1D` and `interp2D` as special cases for the casadi conversion (in the same way that tanh, cosh, etc are all treated as special cases). This is in the class `CasadiConverter`

ferranbrosa, 2020-09-21:
Not sure if this helps, but I had some issues when implementing the drive cycle from data as an interpolant. I ended up using a PyBaMM interpolant, here is an example: https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/change-input-current.ipynb (section Loading in current data)

tinosulzer, 2020-12-29:
Just had a thought for this while doing doing #1310 

In the case where a function `f` expects two children `a` and `b` but we want to provide a form based on interpolating data for `a` while ignoring `b`, we could allow this by specifying this unambiguously at the start of the csv file. For example,

```
###
# Comment block to specify interpolating variable. Do not edit
# Interpolating variable: a
###
```
Then we can use `FunctionParameter.inputs` to read off which variable to use as the interpolant

tinosulzer, 2022-08-02:
I think 2D interpolants now work. This issue can be reopened if something else is needed

closed 2022-08-02 by tinosulzer