pybamm-team/PyBaMM #1102: RecursionError: maximum recursion depth exceeded

vishankkumar, 2020-07-10:
**Describe the bug**
dynamic_plot() function is generating a recursion error that says "RecursionError: maximum recursion depth exceeded while calling a Python object".

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'examples/scripts'
2. Run 'python DFN.py'
3. See error

**Expected behaviour**
Examples are expected to work without error if the prerequisites are met, which I checked earlier.

**Screenshots**
(Screenshot attached)

**Additional context**
Add any other context about the problem here.
[image]

tinosulzer, 2020-07-10:
Looking at the traceback, this looks like a problem with matplotlib rather than pybamm. Do you get this error with the other examples or just this one? Can you try updating to the latest version of matplotlib?

vishankkumar, 2020-07-11:
Thanks for your quick response. Updating matplotlib worked for me, now examples are running fine.

closed 2020-07-11 by vishankkumar