pybamm-team/PyBaMM #1124: ConcretizationTypeError: Abstract tracer value passed to __bool__ for which a concrete value is required.

ibulu, 2020-08-08:
great package! I am trying to use the jaxbdf solver and getting the above error.
minimal code to reproduce this issue:

```
def f(u, t, sigma, rho, beta):
    x, y, z = u
    return jnp.array([sigma * (y - x),
                      x * (rho - z) - y,
                      x * y - beta * z])

u0 = jnp.array([1.0, 0.0, 0.0])
tspan = (0., 10000.)
t = jnp.linspace(0, 100, 1001)

jax_bdf_solver.jax_bdf_integrate(f, u0, t, 10.0, 28.0, 8/3, rtol=1e-8, atol=1e-8)
```
Jax version 0.1.75

martinjrobins, 2020-08-09:
Hi @ibulu, I'm having trouble reproducing your error, I tried the following code (minor alterations from yours) which worked fine for jax=0.1.75 and jaxlib=0.1.52:

```python
import jax
import jax.numpy as jnp
import pybamm
import matplotlib.pyplot as plt

def f(u, t, sigma, rho, beta):
    x, y, z = u
    return jnp.array([sigma * (y - x),
                      x * (rho - z) - y,
                      x * y - beta * z])

u0 = jnp.array([1.0, 0.0, 0.0])
t = jnp.linspace(0, 100, 1001)

y = pybamm.jax_bdf_integrate(f, u0, t, 10.0, 28.0, 8/3, rtol=1e-8, atol=1e-8)
plt.plot(t, y[:, 0])
plt.plot(t, y[:, 1])
plt.plot(t, y[:, 2])
plt.show()
```
Note that it is essential that you return the solution from the `pybamm.jax_bdf_integrate` function, otherwise you will likely get a memory unallocation error from BLAS, for reasons I don't yet understand. However, this isn't the error you are reporting.

Can you please post the full error output so I can see where in `jax_bdf_integrate` it is occuring? This might give some additional clues

ibulu, 2020-08-09:
uninstall and reinstalled pybamm from source and that solved the issue. Thank you for great work.

closed 2020-08-09 by ibulu