pybamm-team/PyBaMM #1135: Quick question

gyouhoc, 2020-09-01:
Hi, 

I am currently working on increasing the capacity (1C discharge capacity) of the model. 
Would you provide me some tips on this? Which material parameter needed to be updated?

tinosulzer, 2020-09-01:
Changing the geometric parameters (height and width of the current collectors) is probably the easiest way. Or you can change the maximum particle concentrations, but then you need to be careful.

gyouhoc, 2020-09-02:
I have fixed geometric parameters.
maximum particle concentrations = Maximum concentration in negative electrode [mol.m-3] and Maximum concentration in positive electrode [mol.m-3] ? 
Which electrode has greater influence on the capacity increase?

tinosulzer, 2020-09-02:
> maximum particle concentrations = Maximum concentration in negative electrode [mol.m-3] and Maximum concentration in positive electrode [mol.m-3] ?

Yes

> Which electrode has greater influence on the capacity increase?

To be honest, this is more of a science question than a software one. We are open for collaboration/research projects on such questions but need to know more information about the system. You can email me vsulzer at umich.edu

closed 2020-09-03 by tinosulzer