pybamm-team/PyBaMM #1143: Add simple equivalent circuit model

Scottmar93, 2020-09-09:
**Summary**
Add a simple equivalent circuit model that is coupled to the "potential pair" current collectors. Will start with just a resistor and add RC circuits later. 

**Reason** 
Simple, easy to parameterize models that can give information about cell utilization. These are good to have in PyBaMM so that the EC and physics-based approaches can be easily compared.

DavidMStraub, 2021-02-10:
Is somebody still pursuing this? I agree it would be great to have this functionality to have a way to compare the easy to parameterize EC models that are used in real-time applications to the physics-based models in the same framework.

tinosulzer, 2021-02-12:
Looks like it doesn't need too much to get this over the line, even if it's just "1D" for now

KennethNwanoro, 2021-02-25:
Is anyone still on this? This is the next thing I would like to implement/see in Pybamm (equivalent circuit models). Useful for comparing physics based models as well as fitting EIS data to decouple internal resistances . If no one is still doing this, I can take it up or work with them. Regards.

PR #2478 (2022-11-29): Issue 1143 equivalent circuit model
# Description

Implementation of Theverin Equivalent Circuit Model (resistor + RC pair). Main things implemented:

- works with the experiment class
- toy parameter set implemented (made up example)
- demonstrates 3D and 2D interpolation working with a model
- variable choice of the number of RC pairs

If anyone has better ideas on where Equivalent circuit models and the associated submodel should be place, would be very helpful. There also is some hacky stuff (e.g. adding additional variables to the model, some additional options) that would also be nice to [... cut]
closed 2022-11-29 by Scottmar93