pybamm-team/PyBaMM #1145: Colorbar on 2d data shows wrong values

TomTranter, 2020-09-10:
**Describe the bug**
I was running a simulation where temperature in a 1+1D model wasn't changing much (less than 1 degree) but the colorbar made it look like it varied by 10s of degrees because it is adjusted up and down by 4% of the absolute value (298) after plotting the mesh when variable_limits="tight". First of all I'm not sure it should be adjusting it at all after plotting and refreshing. Secondly 4% of the data range would make more sense.

**To Reproduce**
Steps to reproduce the behaviour:
```
import pybamm

param = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Marquis2019)
model_1D_lumped = pybamm.lithium_ion.SPMe({"dimensionality": 1,
                                           "current collector": "potential pair",
                                           "thermal": "x-lumped"})
sim_1D_lumped = pybamm.Simulation(model_1D_lumped, parameter_values=param)
sim_1D_lumped.solve([0, 36])
temp = sim_1D_lumped.solution["Cell temperature [K]"].entries
t_min = temp.min()
t_max = temp.max()
variable_limits = {"X-averaged cell temperature [K]": (t_min, t_max),
                   "Cell temperature [K]": (t_min, t_max),
                   "Volume-averaged cell temperature [K]": (t_min, t_max)}
print(t_min, t_max)
sim_1D_lumped.plot(["X-averaged cell temperature [K]",
                    "Cell temperature [K]",
                    "Volume-averaged cell temperature [K]"],
                    variable_limits="tight",
                    )
```

**Expected behaviour**
A true representation of the temperature range that update dynamically for each time step

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

TomTranter, 2020-09-10:
Using the actual variable limits from the dictionary instead gives the best results.
[image]

Not specifying the limits again makes the range too wide but at least shows the right colour and scale
[image]

I am happy to make the changes and adjust the limits based on the data range but also make sure what is shown on the colorbar is the same as the plot - 2 conflated issues here really

tinosulzer, 2020-09-10:
So, just to understand, are you suggesting that the colorbar should be fixed to (min of variable, max of variable) even when `variable_limits` is `"tight"`? Which would be the same behaviour (for 2d data) as having `variable_limits="fixed"`?

tinosulzer, 2020-09-10:
i.e. does the following give you exactly what you want?
```python3
variable_limits = {"X-averaged cell temperature [K]": "tight",
                   "Cell temperature [K]": "fixed",
                   "Volume-averaged cell temperature [K]": "tight"}
```
If so, we could add an option which sets 1D variables to tight and 2D variables to fixed

TomTranter, 2020-09-10:
I think there is a genuine bug. In the first screen shot you see the variation in temperature and corresponding colorbar and they have been scaled differently. ax_min and ax_max increase the range on the colorbar only so my small temperature difference looked as though it was massive.

tinosulzer, 2020-09-10:
Ah yeah, I see

tinosulzer, 2021-02-04:
@TomTranter this might be a good first issue for someone if you don't have time to do and can explain exactly what is needed

TomTranter, 2021-02-04:
Yep, I’d forgotten about this until the other day

tinosulzer, 2021-03-31:
@TomTranter  I wasn't able to reproduce the problem with different scalings. Do you still see it or have we accidentally fixed it?

rtimms, 2023-05-02:
I think this is fixed. @TomTranter can you test and confirm?

closed 2024-10-03 by kratman