pybamm-team/PyBaMM #1163: Build error on Windows CI

tinosulzer, 2020-09-10:
All PRs are failing with this error:
```
ERROR: invocation failed (exit code 1), logfile: D:\a\PyBaMM\PyBaMM\.tox\windows-tests\log\windows-tests-1.log
================================== log start ==================================
Obtaining file:///D:/a/PyBaMM/PyBaMM
    ERROR: Command errored out with exit status 1:
     command: 'D:\a\PyBaMM\PyBaMM\.tox\windows-tests\Scripts\python.EXE' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'D:\\a\\PyBaMM\\PyBaMM\\setup.py'"'"'; __file__='"'"'D:\\a\\PyBaMM\\PyBaMM\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\RUNNER~1\AppData\Local\Temp\pip-pip-egg-info-h8umo9ra'
         cwd: D:\a\PyBaMM\PyBaMM\
    Complete output (10 lines):
    cmake version 3.18.2
    
    CMake suite maintained and supported by Kitware (kitware.com/cmake).
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "D:\a\PyBaMM\PyBaMM\setup.py", line 171, in <module>
        readme = f.read()
      File "C:\hostedtoolcache\windows\Python\3.6.8\x64\lib\encodings\cp1252.py", line 23, in decode
        return codecs.charmap_decode(input,self.errors,decoding_table)[0]
    UnicodeDecodeError: 'charmap' codec can't decode byte 0x90 in position 6688: character maps to <undefined>
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

tlestang, 2020-09-11:
Looks like all-contributors is making edits to the readme that trigger encoding errors on windows

tinosulzer, 2020-09-13:
Ok will close this issue and look into it as part of #1137

closed 2020-09-13 by tinosulzer