pybamm-team/PyBaMM #1172: output data sampling rate

gyouhoc, 2020-09-18:
Hi, 

I have following script to generate a 1sec sample rated output dataset.
        ... 
        timescale = parameter_operation.evaluate(model.timescale)
        sim.specs(parameter_values=parameter_operation)
        sim.solve(t_eval=np.arange(0, 3601, 1),solver=pybamm.CasadiSolver(dt_max=0.001*timescale)) 
  
I loop this script to create multiple output data files.
Mostly, I have a 1sec sample rated output data as intended. 

However, I do have some files as below. 

0.9 sec sample rated output file
[image]

1.1 sec sample rated output file
[image]

Is there any robust way to ensure that I have 1 sec sample rated outputs?

tinosulzer, 2020-09-18:
Can you share the specific conditions which give a 0.9 or 1.1 sec sample rate?

gyouhoc, 2020-09-18:
I don't think why but with another computer and a newer version of pybamm, there is no problem to produce a 1 sec rate with the same condition.  Probably, this is due to the old pybamm version??

closed 2020-09-18 by gyouhoc

tinosulzer, 2020-09-18:
Yeah it could be a new version thing