pybamm-team/PyBaMM #1174: "spherical polar" coordinates ignore left boundary condition

bessman, 2020-09-18:
**Describe the bug**
When using "spherical polar" coordinates, the left boundary condition seems to always be zero flux, regardless of what the user specifies.

**To Reproduce**
Steps to reproduce the behaviour:
1. Run the 'Creating Models/2-a-pde-model.ipynb' notebook.
2. Change the left boundary condition type and/or value to anything.
3. Rerun notebook.
4. Final result is the same.

**Expected behaviour**
The left boundary condition should be user-changeable. Alternatively, if zero flux is intentionally enforced on the left boundary in spherical coordinates, the user should be warned of this fact if they request something else.

tinosulzer, 2020-09-18:
Thanks for pointing out - seems to be something do with with the multiplication by `r**2` which we should be more careful about

ferranbrosa, 2020-09-21:
Hi @bessman! I think that for spherical (and cylindrical coordinates) the condition at r=0 should always be a no-flux condition (note that, in reality, this is an artificial boundary that arises due to the parameterization of the domain). What problem are you trying to solve that gives you a BC other than no-flux?

bessman, 2020-09-21:
I noticed it while working through some of the example notebooks. Specifically, in "Creating Models/2-a-pde-model.ipynb", the left boundary condition type is given as Dirichlet in the code. This should result in an incorrect result, but does not, because the boundary condition is ignored.

As you say, that's not typically going to be a problem when dealing with spherical particels. I just think the behavior should be clearly documented if it is intentional, to avoid surprising the user.

rtimms, 2020-09-21:
thanks @bessman looks like this was a typo in the notebook. as discussed, symmetry always gives no flux at r=0 in spherical polars, but it would be good to add a warning if people try to use a different condition

ferranbrosa, 2020-09-22:
Cool, I will modify the code so it throws an error if a "left" bc is passed that it is not a no-flux condition.

closed 2020-09-22 by tinosulzer

tinosulzer, 2020-09-23:
@all-contributors add @bessman for bugs, examples

tinosulzer, 2020-09-24:
Good bot

PR #1175 (2020-09-22): Issue 1174 raise error bcs sphere other than no-flux
# Description

The code now raises an error if the boundary condition at the origin in a sphere geometry is other than a no-flux condition.

Fixes #1174

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All test [... cut]