pybamm-team/PyBaMM #1176: How to control the size of output if aiming for hundards of cycles

weilongai, 2020-09-22:
**Summary**
Are there options(& documents) to choose the size of output in pybamm?

 **Motivation**
 Why are we doing this?
When cycling for a long time, e.g. hundreds of cycles, it will create a very long array for outpus variables. This may be okay if computer has a big memory. But keeping long arrays will also slow down the simulation. 

What use cases does it support? 
Aim for long cycling

What is the expected outcome?
Options to output results for the specific cycle number, like those experimens doing ICA for every 100 cycle. 
Options to choose which variables to record, and omit others not interesed.
Options to output results ar the user-defined time step. 
Some may be done already, but cannot find a document on this.

tinosulzer, 2020-09-22:
Some of this functionality already exists, some doesn't. Take a look at the notebook `solution-data-and-processed-variables.ipynb`

- Only record some variables: this already is the case. When you solve a model, all it calculates is the states `y` for solving the actual equations. All of the other variables, which are calculated from these states `y` (such as the voltage) are only created at the point when you call them from the solution dictionary.
- Output results at the user-defined time step: this functionality exists, e.g. `solution["Terminal voltage [V]"](tspan)` returns the voltage at times `tspan`
- Specific cycle number: this doesn't exist but should be added. The difficulty is that it's hard to know what "one cycle" is, for example if you do 
```python3
pybamm.Experiment(
    [
        "Discharge at C/10 for 10 hours or until 3.3 V",
        "Rest for 1 hour",
        "Charge at 1 A until 4.1 V",
        "Hold at 4.1 V until 50 mA",
        "Rest for 1 hour",
        "Discharge at C/10 for 10 hours or until 3.3 V",
        "Rest for 1 hour",
        "Charge at 1 A until 4.1 V",
        "Hold at 4.1 V until 50 mA",
        "Rest for 1 hour"
    ]
)
```
then how do you know which instructions are one cycle? Right now there is something called "sub solutions" which returns the solution associated with one step of an experiment (check out `cccv.py`). For example in this case `solution.sub_solutions[2]["Terminal voltage [V]"].data would return the voltage for the  "Charge at 1 A until 4.1 V" step. 
We could change the experiment class to accept something of the form
```python3
pybamm.Experiment(
    [
        ("Discharge at C/10 for 10 hours or until 3.3 V",
        "Rest for 1 hour",
        "Charge at 1 A until 4.1 V",
        "Hold at 4.1 V until 50 mA",
        "Rest for 1 hour"),
        ("Discharge at C/10 for 10 hours or until 3.3 V",
        "Rest for 1 hour",
        "Charge at 1 A until 4.1 V",
        "Hold at 4.1 V until 50 mA",
        "Rest for 1 hour")
    ]
)
```
where the brackets make it clear what is one cycle, and then using that it should be possible to separate cycles

weilongai, 2020-09-23:
Thanks for your reply. If only `y` is being recorded and updated in a simulation, there would be no memory issue if the number 
 of unknowns is not large and other variables can be extracted from `y`. 
- I'm just curious if it's possible to control the size of `y` as well, for extreme working conditions. For instance, the solver solves at every 0.1s but records at every 10s for a long cycling time, e.g. 10^7s. 
- There are some experiments using the below commands, where the ones being multiplied are assigned as one cycle. When it has this definition, it will be easier to add outputs like charge and discharge capacities per cycle for cycling tests. 
```
total_cycles=20
experiment = pybamm.Experiment(
    [
        "Discharge at 1C until 2.5 V",
        "Rest for 600 seconds",
        "Charge at 1C until 4.2 V",
        "Hold at 4.2 V until 1 mA",
    ] * total_cycles
)
```

tinosulzer, 2020-09-23:
Right now there is no way to control the size of `y`. I guess we could add some functionality to do this in the post-processing. You should be able to do something manually though.

The way python lists work means that it doesn't know how the list was created. For example, `["a"] * 3 + ["b"] * 2` is exactly the same as `["a", "a", "a", "b", "b"]`. So you would need explicit sublists or tuples

rtimms, 2020-09-24:
> I'm just curious if it's possible to control the size of y as well, for extreme working conditions. For instance, the solver solves at every 0.1s but records at every 10s for a long cycling time, e.g. 10^7s.

re this: the array `t_eval` that you pass to the solver is the times you want it to record at, but the solver does adaptive time-stepping to decide what steps to actually take. you can control the solver steps via the tolerances and maximum step size (e.g. see the casadi docs here http://casadi.sourceforge.net/v2.0.0/api/html/db/d3d/classcasadi_1_1Integrator.html). different solvers will handle the time-stepping differently, so you'll need to look at the documentation for full details. for experiments the frequency at which the solution is recorded at is set by the `period`

weilongai, 2020-09-24:
Thanks all for your reply, which are helpfull. There is no issue now.

closed 2020-09-24 by weilongai