pybamm-team/PyBaMM #1182: Tag PyBaMM citations

ferranbrosa, 2020-09-30:
**Summary**
The way citations are printed now (all together) can make it quite hard for an inexperienced user to know why the need to cite each reference. It would be good to add a tag when registering PyBaMM citations and also some extra functionality that, when printing the citations, reads the tags and groups them in a more meaningful way. For example, we could print some sort of list or table that lists:

- Model: Fuller1994
- Parameters: Chen2020
- Solver: Andersson 2019
and so on...

Please add your comments and suggestions

 **Motivation**
 Make it easier for new or inexperienced users to know what's the purpose of each reference, which might lead in them just dumping the citations in their papers all together. Adding this functionality should lead to higher quality citations in the papers.

rtimms, 2020-09-30:
I think this sounds great! At the moment the result of `print_citations()` can be a bit verbose and it's not super readable. Maybe it would be good to print a table/list like you suggest and then generate a .bib file (it looks like you can do this by passing a filename, but perhaps it should print something in `cwd` if no filename is passed). Then we could print the table/list like you have above along with a message saying a bib file as been generated. If its possible to automate, you could also add something the "note" field of the bibtex entry explaining where the citation was used.

ferranbrosa, 2020-10-01:
Maybe we could rename the current `print` method to `print_to_bibtex`, as I think it is a useful feature for people using LaTeX. 

For those who don't, maybe we could use a more user friendly function. I think the list/table would be useful to list the tags, but then we should provide them with the full reference somewhere. Options that I can think of:

- Point them to the documentation.
- Print formatted citation from the Bibtex file.

Thoughts?

rtimms, 2020-10-01:
Is there an easy way to automatically print the citation in a nice way from a bibtex file? If so, it would be nice to generate a pdf or similar with all the citations and what they are used for. Maybe it could also print some other useful info like current version number etc. so that it also serves as a helpful document for reproducing results. I like the option of printing to bibtex, but maybe you are right - it isn't most useful for everyone.

ferranbrosa, 2020-10-01:
There a few packages (see [this Stackoverflow post](https://stackoverflow.com/questions/30768745/is-there-a-reliable-python-library-for-taking-a-bibtex-entry-and-outputting-it-i)) but it seems that none is great. Maybe the closest one is Pybtex but doesn't seem to be that active either...

brosaplanella, 2020-12-18:
Just took another look and there has been a new release of [Pybtex](https://pybtex.org/). In my opinion, this would be the best way to render the references nicely.

Does this sound like a good idea or would it be better to not add another dependency for such a minor thing?

Saransh-cpp, 2021-02-05:
I was following this and the mentioned issue, are you looking to create a new method in Citations class to print citations in a better way? The print method (which you are thinking of renaming) already uses ```pybtex``` as described in the #1330 . 

And as I was exploring the citations section,  ```pybamm.citations.register("Sulzer2020")``` gives me a key error. I followed ahead to check, the read_citations method reads all the citations and writes it down in a map so I tried running that method first but still the same error. How can I fix this? and I'll be happy to help and solve this issue.

rtimms, 2021-02-05:
The goal of this issue was to somehow take citations with where they are used in the code (e.g. are they used in a solver or a particular mode) so that users know why they need to cite that paper. I think we are happy with how the citations are printed now that we are using `pybtex`.

Some of the reference keys were updated in #1330 to make them all consistent - are you up to date with the latest develop branch? This might be the cause of the key error.

Saransh-cpp, 2021-02-05:
Yes, after the pulling the branch it works, thank you.

ayeankit, 2023-02-20:
Hello @tinosulzer  and  @tinosulzer , I want to work on this issue but I am new to this repo and I need help. Please assign me this.

rtimms, 2023-02-20:
At the moment we register citations (e.g. [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_ion/dfn.py#L37)) in various parts of the code, which then get logged using the `Citations` class, see [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/citations.py). You can then print out which papers you need to cite based on what code you used. 

It would be helpful to also log somehow _where_ the citations were registered. Probably best in some high-level way like by the folder inside `pybamm` that the code lives in (`models`, `solvers`, `plotting`, etc.). This could show up as a note in the bibtex entry, or we could print citations organised by folder.

agriyakhetarpal, 2023-04-01:
Hi @rtimms, I'm not sure how a table would be implemented but perhaps a modification or improvement for the `pybamm.citations.register()` method could be to add another argument (or other functionality) to return the `__qualname__` of the class it was called from, such that it can log where the citations were registered like you suggest
```python
def(register, key, cited_from):
# additional logic
```

and then after calling `pybamm.citations.register()`, say, for the DFN lithium-ion model:
```python
>>> pybamm.print_citations()
```
returns
```
[6] Scott G. Marquis, Valentin Sulzer, Robert Timms, Colin P. Please, and S. Jon Chapman. An asymptotic derivation of a single particle model with electrolyte. Journal of The Electrochemical Society, 166(15):A3693–A3706, 2019. doi:10.1149[/2.0341915jes.](https://file+.vscode-resource.vscode-cdn.net/2.0341915jes.)

cited due to the use of pybamm.lithium_ion.DFN()
```
Another way to do the same is to wrap the `register()` function in a class decorator `@register_citation` with a custom metaclass to alter the `__call__` method which can then be called whenever a model is initialised. I am happy to work on this or any suggestions you may have (provided @ayeankit gives me a go-ahead, since they are assigned to this issue currently).

ayeankit, 2023-04-02:
@agriyakhetarpal go ahead, currently I working on different issues.

rtimms, 2023-04-03:
@agriyakhetarpal sounds great, thanks!

tinosulzer, 2023-05-01:
Re-opening as discussed in #2908

PR #2862 (2023-04-13): Issue 1182 verbose outputs and tagging PyBaMM citations
# Description

Adds a caller method for `citations.py` to tag PyBaMM citations to their use of particular models or solvers and adds verbose output while printing citations

Fixes #1182 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (b [... cut]
PR #2961 (2023-06-09): Issue 1182 faster tagging for PyBaMM citations 
# Description

Enables a faster method to tag citations based on where they come up in the code during a simulation (models, solvers, etc.) – via a `caller_name` method that calls `sys._getframe()` to access current frames for the `pybamm.citations.register()` function. 

This PR continues the discussion on #2862 and #2908, with some results of the speedup available at https://github.com/pybamm-team/PyBaMM/issues/2908#issuecomment-1537077212. Here, `sys._getframe` is preferred to `inspect._getframe()` for avoiding the internal `hasattr()`, in ord [... cut]
closed 2023-06-09 by tinosulzer