pybamm-team/PyBaMM #1195: Track cycles in solution

tinosulzer, 2020-10-17:
Change experiment and solution classes so that `solution.cycles` is a list of the cycles (e.g. 1 CCCV charge + CC discharge)
See #1176

jkoeller, 2020-12-16:
A related functionality that would be nice is the ability to stop a simulation after capacity drops below e.g. 80%. Once the notion of cycle is defined, the discharge capacity for that cycle could be computed.

This is not the sort of thing that would be handled by an `Event`, correct? `Event`s get evaluated at each time step? Rather, this would be some new `if` statement with a `break` evaluated [here](https://github.com/pybamm-team/PyBaMM/blob/96029c09ce835220838ef404ade3521d4f479e07/pybamm/simulation.py#L475) perhaps?

tinosulzer, 2020-12-16:
Good idea, yes that would be the place to implement it, and it is separate from `Event`.
Since it is an "event" for the experiment, the 80% cut-off could simply be specified when creating the experiment class.
More generally, we should have "summary" variables for each "cycle" in a solution, which are a function of "cycle number" but not time within each solution, such as "Discharge capacity [A.h]" but also some averaged measure of LAM, LLI, etc.

I think Jack Lonnborn (@lonnbornj ?) has made a start on this issue, but don't know how far he has got

lonnbornj, 2020-12-17:
I have implemented the first part of this (definition of cycles by the user on creation of an Experiment), and currently working on extracting the cycles again at around that line in `simulation.py`. I'll keep the cut-off functionality in mind as I do so.

PR #1309 (2020-12-31): Issue 1195 track cycles in solution
# Description

Altered the Experiment class to parse experiments expressed in terms of cycles (in the form of tuples of operating conditions). The Simulation class creates `solution.cycles` -- a list of cycles. Each cycle is a tuple of `sub_solution` objects.

Fixes #1195 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (n [... cut]
closed 2020-12-31 by tinosulzer