pybamm-team/PyBaMM #1207: surface form for SPMe

rtimms, 2020-10-22:
At the moment if you pass the option `{"surface form":"differential"}` or `{"surface form": "algebraic"}` to SPMe nothing happens. SPMe always uses `pybamm.electrolyte_conductivity.Composite`. 

We should either raise an `OptionError` or add the relevant surface models for SPMe

brosaplanella, 2020-10-23:
I tried implementing this in #884 in case it was a quick fix but I am getting errors for both models (full and leading-order). For the moment I set it to throw a `NotImplementedError` so if that is enough we may close this issue once #884 is merged.

Otherwise, if we want to implement the model, we need to derive the appropriate surface form for `Composite`. Is there any reference on this kind of formulation?

tinosulzer, 2020-10-23:
There is my thesis, if you are feeling brave, but I wouldn't go down that route. In theory the leading order form should work though. What was the error?

brosaplanella, 2020-10-24:
The differential works but the algebraic doesn't. Here is the error:
```
======================================================================
ERROR: test_surface_form_algebraic (__main__.TestSPMe)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/ferranbrosa/PyBaMM/tests/unit/test_models/test_full_battery_models/test_lithium_ion/test_spme.py", line 119, in test_surface_form_algebraic
    model.check_well_posedness()
  File "/home/ferranbrosa/PyBaMM/pybamm/models/base_model.py", line 377, in check_well_posedness
    self.check_well_determined(post_discretisation)
  File "/home/ferranbrosa/PyBaMM/pybamm/models/base_model.py", line 469, in check_well_determined
    raise pybamm.ModelError("model is overdetermined (repeated keys)")
pybamm.expression_tree.exceptions.ModelError: model is overdetermined (repeated keys)

----------------------------------------------------------------------
```

PR #1257 (2020-11-26): #1207 add composite surface form
# Description
Add composite surface form

Fixes #1207

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] A [... cut]
closed 2020-11-26 by rtimms