pybamm-team/PyBaMM #1211: Review of CONTRIBUTING guidelines

dalonsoa, 2020-10-23:
The [CONTRIBUTING guidelines](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md) have several points that are either incomplete or outdated. These are the things I've noticed:

- We use `tox`, but nowhere it is said that it needs to be installed or how to do it. Indeed, there's no brief discussion about we using `tox` as the development environment. 
- Maybe some discussion on how to setup the virtual environment created by `tox` in VSCode, also working with the debugger. 
- In the [Installation](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installation) section, it is said that `tox -e dev` installs all development options, however the Sundials solver is not installed.
- About Sundials, there's no mention in that file at all.  I guess it should be installed with `tox -e pybamm-requires`, right? 
- If you activate the `tox` environment, within it there's no `tox` command so any subsequent `tox -e {whatever}` will fail. 
- Travis CI is no longer used as the CI system. It should be removed [from here](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#travis-ci) and mention GitHub Actions instead.
- In general, the file might benefit form some content reorganisation, but that's just a purely subjective comment. 

I'm happy to tackle these issues myself and open a PR with the changes, but wanted to check with you first and gather any other changes that needs to be done.

tinosulzer, 2020-10-23:
Agree with all these suggestions, thanks @dalonsoa !

> We use tox, but nowhere it is said that it needs to be installed or how to do it. Indeed, there's no brief discussion about we using tox as the development environment

👍 

> Maybe some discussion on how to setup the virtual environment created by tox in VSCode, also working with the debugger.

Yeah it would be nice to give a full workthrough of how to set up with VS Code for those who have never used python and are used to e.g. the MATLAB IDE. If people prefer something else (vim, spyder, pycharm) then they can ignore that section

> In the Installation section, it is said that tox -e dev installs all development options, however the Sundials solver is not installed.
About Sundials, there's no mention in that file at all. I guess it should be installed with tox -e pybamm-requires, right?

Is it possible to install sundials through tox? I thought it required separate steps @tlestang 

> If you activate the tox environment, within it there's no tox command so any subsequent tox -e {whatever} will fail.

So should tox be a dependency so it gets installed inside the env?

> Travis CI is no longer used as the CI system. It should be removed from here and mention GitHub Actions instead.

👍 

> In general, the file might benefit form some content reorganisation, but that's just a purely subjective comment.

👍

tlestang, 2020-10-26:
Thanks Diego-  indeed it looks like the guidelines are not up to date with the latest tox workflow..

> Is it possible to install sundials through tox?

Yes, with `tox -e pybamm-requires` which downloads and compiles Sundials. This is intended to be used on GNU/Linux platforms only: on MacOS, the recommended approach is to install sundials through homebrew as that's easier and more reliable. Last time I checked, versions of Sundials available in most GNU/Linux distros were too old so it must be built "manually". I think this is in the installation docs, but should also be in the contributing guidelines - or at the very least they should include a link to the installation docs.

> So should tox be a dependency so it gets installed inside the env?

That feels a  bit redundant, maybe the way I designed the Tox workflow isn't great - I'm not sure at this time if installing tox in the development environment is common practice.
@dalonsoa do you have experience whith Tox?

dalonsoa, 2020-10-27:
I hadn't heard of `tox` until you used it here, so I've no idea of what's the normal use. I also think it is redundant, but I was mentioning the issue because approach described in the file is just not working. I think the user should either:

- Activate the `tox` environment created with `tox -e dev` and then run the tests directly without invoking `tox` (eg. `python run-tests.py --unit --folder all`), or
- Do not activate the `tox` environment and then run the tests using `tox` (eg. `tox -e quick`). 

The later has the limitation that if the user wants to run the examples or run their own calculations, they cannot: they need to create and activate an environment, anyway.

tlestang, 2020-10-28:
I'd actually go with the second option. 
Development work as well as running calculations/examples is done inside the `tox` environment.
Running the tests and building the documentation is done from outside this environment using `tox -e tests`, `tox -e docs`.

It realise it sounds odd to deactivate the environment before running the tests - but an advantage of `tox` is that it will create a new environment dedicated to the task at hand, isolated from the environment `tox` was invoked from (for instance only a small number of environment variables are inherited by the test environment). The tests would not have been ran from the development environment anyway. Does this make sense?

dalonsoa, 2020-10-28:
It might be a bit confusing when you are using an IDE that automatically activates the environment (VSCode, PyCharm) to get out of the environment to run the tests, though. Being that the case, I think it makes sense to install tox within the environment to benefit from both, being inside an environment for development and running examples and simulations and exploiting tox functionality and compartmentalization of environments for different tasks.

tlestang, 2020-10-28:
That's a good point. Okay - let's add tox to the dependencies for the `dev` testenv then. 

Just thought of a third approach: we could recommend the use of [pipx](https://github.com/pipxproject/pipx) to install `tox`. This allows the installation of python applications (tox, black, flake8, pylint...) in separate environments. For instance `tox`  would be available  from the `dev` env even if it's not installed in it. haven't tried it yet, but it looks widely used. 
Here's a practical introduction to it: https://pycon.switowski.com/02-packages/pipx/

dalonsoa, 2020-10-28:
Mmmm… looks like an interesting tool for these kind of things (using it for black and flake8 is quite compelling!). Maybe we can give it a try ourselves and then decide?

There’s going to be an explosion of environments whenever something is run…

dalonsoa, 2020-10-29:
I've checked `pipx` this and it works quite nicely, to be honest. I would include this as a suggestion, though. In the end, it is yet another tool that is not directly used by PyBaMM, but rather by a tool actually used by PyBaMM. 

In summary, I'd include the following options and let the user decide:

1. Install `tox` normally with `pip` (or `conda`), and then, later on, install `tox` within the environment.
2. Install `tox` with `pipx` so it is available system wide. 

Both options results in the possibility of running things like `tox -e quick` with the environment activated.

What do you think?

tlestang, 2020-10-30:
That sounds good to me! If users don't install `tox` in the dev environment, they can still run the tests with `pyhon run-tests` anyway.

Saransh-cpp, 2021-02-03:
> The [CONTRIBUTING guidelines](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md) have several points that are either incomplete or outdated. These are the things I've noticed:
>
> * We [...]

Yes I agree, initially I wasn't able to set up the virtual environment because of 'tox' in my commands which I had no idea that I had to install seperately, then I decided to run pybamm without creating a virtual environment and it worked without any problems. If you need any kind of help, I'll be happy to contribute and make the documentation (which is quite good right now) better for complete newbies.

tinosulzer, 2021-02-03:
That would be great. Feel free to open issues and/or PRs for anything you find that is missing

dalonsoa, 2021-02-04:
I am really sorry I had this abandoned. I was halfway the rewriting but other things came up. If you want to take on this, please go ahead. It will take me at least 2 more weeks to be able to go back to this issue.

Saransh-cpp, 2021-02-04:
Okay! I'll start working on the changes.

Saransh-cpp, 2021-02-04:
I have been experimenting with all the installation steps, I think they should go like this - 

1. Installation without any virtual environment
a) forking and cloning the repository and installing the necessary dependencies.

2. Creating a virtual environment (Here 2 options as discussed above, using pip and pipx)
a) ```pip/pipx install tox``` 
b) forking and cloning the repository
c) ```python -m tox -e windows-dev```
d)```.tox\windows-dev\Scripts\activate.bat```
e)```pip install tox``` in the virtual env ( not required with pipx )

And lastly adding the link to your docs which explains installing sundial using homebrew in mac.

I'll update everything around these steps, should I make any changes in the above steps ( this is just an overview, I'll make them more clear in the actual file )?

PR #1361 (2021-02-09): Docs: Improved CONTRIBUTIONS.md
# Description

Made some changes in CONTRIBUTIONS.md, making it easier for complete newbies to read as mentioned in #1211. Please review and let me know if you want any changes. Thank you.

Fixes #1211 

## Type of change

Change in documentation, more precisely in CONTRIBUTIONS.md.
How should I update the CHANGELOG.md in case of documentation change? Please guide.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist [... cut]
closed 2021-02-09 by tlestang