pybamm-team/PyBaMM #1215: CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]

gyouhoc, 2020-10-27:
Hi, 

I have following solver error. Would you help me to solve this issue? 

=code=

model = pybamm.lithium_ion.DFN({"thermal": "x-full"})
timescale = parameter_values.evaluate(model.timescale)
sim = pybamm.Simulation(model,parameter_values=parameter_values)
sim.solve(t_eval=np.arange(0, 5000, 1),solver=pybamm.CasadiSolver(mode="safe", rtol=1e-9, atol=1e-9))
sim.plot()

=== error message == 

CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
Newton/Linesearch algorithm failed to converge.
Newton/Linesearch algorithm failed to converge.
At t = 1.48116e-006, , mxstep steps taken before reaching tout.
At t = 1.48116e-006, , mxstep steps taken before reaching tout.
At t = 1.48116e-006, , mxstep steps taken before reaching tout.
---------------------------------------------------------------------------
SolverError                               Traceback (most recent call last)
<ipython-input-54-eb18faf4855d> in <module>
      2 timescale = parameter_values.evaluate(model.timescale)
      3 sim = pybamm.Simulation(model,parameter_values=parameter_values)
----> 4 sim.solve(t_eval=np.arange(0, 5000, 1),solver=pybamm.CasadiSolver(mode="safe", rtol=1e-9, atol=1e-9))
      5 sim.plot()

~\Documents\PyBaMM\pybamm\simulation.py in solve(self, t_eval, solver, external_variables, inputs, check_model)
    416 
    417             self.t_eval = t_eval
--> 418             self._solution = solver.solve(self.built_model, t_eval, inputs=inputs)
    419 
    420         elif self.operating_mode == "with experiment":

~\Documents\PyBaMM\pybamm\solvers\base_solver.py in solve(self, model, t_eval, external_variables, inputs)
    573             timer.reset()
    574             new_solution = self._integrate(
--> 575                 model, t_eval_dimensionless[start_index:end_index], ext_and_inputs
    576             )
    577             new_solution.solve_time = timer.time()

~\Documents\PyBaMM\pybamm\solvers\casadi_solver.py in _integrate(self, model, t_eval, inputs)
    191                             solving the model up to this time only or reducing dt_max.
    192                             """.format(
--> 193                                 t
    194                             )
    195                         )

SolverError: 
                            Maximum number of decreased steps occurred at t=0.0. Try
                            solving the model up to this time only or reducing dt_max.

rtimms, 2020-10-27:
Hi @gyouhoc 

Can I ask what parameter values you are using? Running the below works ok

```python3
import pybamm
import numpy as np

pybamm.set_logging_level("INFO")

model = pybamm.lithium_ion.DFN({"thermal": "x-full"})
parameter_values = model.default_parameter_values
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve(
    t_eval=np.arange(0, 5000, 1),
    solver=pybamm.CasadiSolver(mode="safe", rtol=1e-9, atol=1e-9),
)
sim.plot()
```

but you do get a `Newton/Linesearch algorithm failed to converge.` warning. The model actually only makes it up to 3654s before hitting the voltage cutoff (you can check this by looking at `sim.solution["Time [s]"].entries `). The warning comes from where the solver tries to step past the event and fails. This step is then rejected and a shorter internal time step is made.

My first guess was that what is happening is that you are hitting some event, but because the tolerances are tight the solver is trying hard to hone in on the event time and taking more steps than the maximum allowed (you can change the max steps as an option). However, it looks like it fails at t=0, so maybe this isn't the case...

Could you try integrating for a shorter time, even just a few seconds, to see if it makes it past the first step?

gyouhoc, 2020-10-27:
These are my parameter values. 

{'1 + dlnf/dlnc': 1.0,
 'Activation_energy_electrolyte': 34700,
 'Ambient temperature [K]': 298.15,
 'Cation transference number': 0.2594,
 'Cell capacity [A.h]': 5.0,
 'Current function [A]': 5.0,
 'DESIGN_ANODE_DENSITY_HOST': 2266,
 'DESIGN_ANODE_MOLWT_HOST': 0.072064,
 'DESIGN_ANODE_NLI_HOST': 0.727,
 'DESIGN_ANODE_STOICHMAX': 1,
 'DESIGN_ANODE_VF_IM': 0.0351,
 'DESIGN_CATHODE_DENSITY_HOST': 4800,
 'DESIGN_CATHODE_MOLWT_HOST': 0.088945,
 'DESIGN_CATHODE_NLI_HOST': 0.75,
 'DESIGN_CATHODE_STOICHMAX': 1,
 'DESIGN_CATHODE_STOICHMIN': 0.398,
 'DESIGN_CATHODE_VF_IM': 0.125094,
 'DESIGN_CELL_ICL_FRAC_CATUSE': 0.06,
 'DESIGN_CELL_NP_RATIO': 1.153,
 'Edge heat transfer coefficient [W.m-2.K-1]': 0.3,
 'Electrode height [m]': 0.089,
 'Electrode width [m]': 2.34,
 'Electrolyte conductivity [S.m-1]': <function electrolyte_conductivity_Nyman2008 at 0x000001A208841048>,
 'Electrolyte diffusivity [m2.s-1]': <function electrolyte_diffusivity_Nyman2008 at 0x000001A207832E18>,
 'Electrolyte_diffusion_activation_energy': 37040,
 'Initial concentration in electrolyte [mol.m-3]': 1200.0,
 'Initial concentration in negative electrode [mol.m-3]': 18081.0,
 'Initial concentration in positive electrode [mol.m-3]': 20090.0,
 'Initial temperature [K]': 298.15,
 'Lower voltage cut-off [V]': 2.5,
 'Maximum concentration in negative electrode [mol.m-3]': 25000.0,
 'Maximum concentration in positive electrode [mol.m-3]': 40500.0,
 'Negative current collector conductivity [S.m-1]': 58411000.0,
 'Negative current collector density [kg.m-3]': 7925.311203,
 'Negative current collector specific heat capacity [J.kg-1.K-1]': 385.0,
 'Negative current collector surface heat transfer coefficient [W.m-2.K-1]': 0.0,
 'Negative current collector thermal conductivity [W.m-1.K-1]': 401.0,
 'Negative current collector thickness [m]': 9.1e-06,
 'Negative electrode Bruggeman coefficient (electrode)': 1.5,
 'Negative electrode Bruggeman coefficient (electrolyte)': 1.5,
 'Negative electrode OCP [V]': <function graphite_ocp_NCM2020 at 0x000001A207832730>,
 'Negative electrode OCP entropic change [V.K-1]': 0.0,
 'Negative electrode active material volume fraction': 0.727,
 'Negative electrode cation signed stoichiometry': -1.0,
 'Negative electrode charge transfer coefficient': 0.5,
 'Negative electrode conductivity [S.m-1]': 100.0,
 'Negative electrode density [kg.m-3]': 1875.0,
 'Negative electrode diffusivity [m2.s-1]': <function graphite_diffusivity_NCM2020_peter at 0x000001A2078328C8>,
 'Negative electrode double-layer capacity [F.m-2]': 0.2,
 'Negative electrode electrons in reaction': 1.0,
 'Negative electrode exchange-current density [A.m-2]': <function graphite_LGM50_electrolyte_exchange_current_density_Chen2020 at 0x000001A207832598>,
 'Negative electrode porosity': 0.39,
 'Negative electrode specific heat capacity [J.kg-1.K-1]': 700.0,
 'Negative electrode surface area to volume ratio [m-1]': 345000.0,
 'Negative electrode thermal conductivity [W.m-1.K-1]': 1.137,
 'Negative electrode thickness [m]': 4.46e-05,
 'Negative particle distribution in x': 1.0,
 'Negative particle radius [m]': 5e-06,
 'Negative tab centre y-coordinate [m]': 0.098,
 'Negative tab centre z-coordinate [m]': 1.235,
 'Negative tab heat transfer coefficient [W.m-2.K-1]': 10.0,
 'Negative tab width [m]': 0.13,
 'Number of cells connected in series to make a battery': 1.0,
 'Number of electrodes connected in parallel to make a cell': 1.0,
 'Positive current collector conductivity [S.m-1]': 36914000.0,
 'Positive current collector density [kg.m-3]': 2679.2452829999997,
 'Positive current collector specific heat capacity [J.kg-1.K-1]': 897.0,
 'Positive current collector surface heat transfer coefficient [W.m-2.K-1]': 0.0,
[... 56 lines cut ...]

parameter_values["DESIGN_CATHODE_STOICHMAX"] = 1 - parameter_values["DESIGN_CELL_ICL_FRAC_CATUSE"] 

#parameter_values["Negative electrode active material volume fraction"] = 0.8
#parameter_values["Positive electrode active material volume fraction"] = 0.8
parameter_values["Positive electrode porosity"] = 0.416
parameter_values["Negative electrode porosity"] = 0.39

parameter_values["Positive electrode active material volume fraction"] = 1-parameter_values["Positive electrode porosity"] - parameter_values["DESIGN_CATHODE_VF_IM"]
parameter_values["Positive electrode surface area to volume ratio [m-1]"] = 3 * parameter_values["Positive electrode active material volume fraction"] /parameter_values["Positive particle radius [m]"]

parameter_values["DESIGN_ANODE_STOICHMAX"] = (1/parameter_values["DESIGN_CELL_NP_RATIO"])*((parameter_values["DESIGN_CATHODE_STOICHMAX"]-parameter_values["DESIGN_CATHODE_STOICHMIN"])/(1-parameter_values["DESIGN_CATHODE_STOICHMIN"]))

num = parameter_values["Positive electrode thickness [m]"]*parameter_values["Maximum concentration in positive electrode [mol.m-3]"]*(1-parameter_values["DESIGN_CATHODE_STOICHMIN"])
den = parameter_values["Negative electrode thickness [m]"]*parameter_values["Maximum concentration in negative electrode [mol.m-3]"]*(1-0)

parameter_values["Negative electrode active material volume fraction"] = parameter_values["Positive electrode active material volume fraction"]*parameter_values["DESIGN_CELL_NP_RATIO"]*(num/den) 
parameter_values["Negative electrode porosity"] = 1-parameter_values["Negative electrode active material volume fraction"] - parameter_values["DESIGN_ANODE_VF_IM"]
[... cut ...]
#parameter_values["Positive electrode surface area to volume ratio [m-1]"] = 3 * parameter_values["Positive electrode active material volume fraction"] /parameter_values["Positive particle radius [m]"]
parameter_values["Negative electrode surface area to volume ratio [m-1]"] = 3 * parameter_values["Negative electrode active material volume fraction"] /parameter_values["Negative particle radius [m]"]

parameter_values["Initial concentration in negative electrode [mol.m-3]"] = parameter_values["DESIGN_ANODE_STOICHMAX"]+0.01
parameter_values["Initial concentration in positive electrode [mol.m-3]"] = parameter_values["DESIGN_CATHODE_STOICHMIN"]-0.01

rtimms, 2020-10-27:
did integrating for a shorter time work? would be good to find out if one of the parameter values is responsible. looks like you end up with something like a divide by 0 somewhere

gyouhoc, 2020-10-27:
keep looping :) 

==code==

model = pybamm.lithium_ion.DFN({"thermal": "x-full"})
timescale = parameter_values.evaluate(model.timescale)
sim = pybamm.Simulation(model,parameter_values=parameter_values)
sim.solve(t_eval=np.arange(0, 0.5, 0.1),solver=pybamm.CasadiSolver(dt_max=0.1))
sim.plot()

==warning==

CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 20, col 0).") [.../casadi/core/oracle_function.cpp:265]
CasADi -  WARNING("roots:g failed: Inf detected for output x, at (row 0, col 0).") [.../casadi/core/oracle_function.cpp:265]
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
At t = 1.49921e-006 and h = 4.02258e-017, the corrector convergence failed repeatedly or with |h| = hmin.
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
[... cut ...]
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed
At t = 1.49989e-006 and h = 6.37792e-018, the corrector convergence failed repeatedly or with |h| = hmin.
​
​

rtimms, 2020-10-27:
ok, seems like it might be to do with one of the parameters. have you been able to run using these parameters before? you could try running without thermal to help narrow down which parameters are responsible. you could also try simpler models (e.g. try SPM to see if the problem is in the electrolyte parameters)

gyouhoc, 2020-10-27:
the updated parameters are new. 

I tried following code but I still have a same issue. 

==Code ==

model = pybamm.lithium_ion.DFN()
timescale = parameter_values.evaluate(model.timescale)
sim = pybamm.Simulation(model,parameter_values=parameter_values)
sim.solve(t_eval=np.arange(0, 0.5, 0.1),solver=pybamm.CasadiSolver(dt_max=0.1))
sim.plot()

rtimms, 2020-10-27:
does the code run with the first dictionary of parameter values you posted, before the updates? if yes, then you could try just updating a few at a time to see when it breaks

gyouhoc, 2020-10-27:
when I use SPM, it seems like I am able to solve the equations. 
I will find the parameter that breaks the simulation as you describe. 

==Code == 

model = pybamm.lithium_ion.SPM()
timescale = parameter_values.evaluate(model.timescale)
sim = pybamm.Simulation(model,parameter_values=parameter_values)
sim.solve(t_eval=np.arange(0, 3700, 1),solver=pybamm.CasadiSolver(dt_max=0.1))
sim.plot()

==Error==

---------------------------------------------------------------------------
SolverError                               Traceback (most recent call last)
<ipython-input-73-96a599fae00a> in <module>
      3 sim = pybamm.Simulation(model,parameter_values=parameter_values)
      4 sim.solve(t_eval=np.arange(0, 3700, 1),solver=pybamm.CasadiSolver(dt_max=0.1))
----> 5 sim.plot()

~\Documents\PyBaMM\pybamm\simulation.py in plot(self, quick_plot_vars, testing)
    555 
    556         self.quick_plot = pybamm.dynamic_plot(
--> 557             self._solution, output_variables=quick_plot_vars, testing=testing
    558         )
    559 

~\Documents\PyBaMM\pybamm\quick_plot.py in dynamic_plot(*args, **kwargs)
     58     """
     59     kwargs_for_class = {k: v for k, v in kwargs.items() if k != "testing"}
---> 60     plot = pybamm.QuickPlot(*args, **kwargs_for_class)
     61     plot.dynamic_plot(kwargs.get("testing", False))
     62     return plot

~\Documents\PyBaMM\pybamm\quick_plot.py in __init__(self, solutions, output_variables, labels, colors, linestyles, figsize, time_unit, spatial_unit, variable_limits)
    270                     )
    271 
--> 272         self.set_output_variables(output_variable_tuples, solutions)
    273         self.reset_axis()
    274 

~\Documents\PyBaMM\pybamm\quick_plot.py in set_output_variables(self, output_variables, solutions)
    297                 variables[i] = []
    298                 for var in variable_tuple:
--> 299                     sol = solution[var]
    300                     # Check variable isn't all-nan
    301                     if np.all(np.isnan(sol.entries)):

~\Documents\PyBaMM\pybamm\solvers\solution.py in __getitem__(self, key)
    195         else:
    196             # otherwise create it, save it and then return it
--> 197             self.update(key)
    198             return self._variables[key]
    199 

~\Documents\PyBaMM\pybamm\solvers\solution.py in update(self, variables)
    163             else:
    164                 var = pybamm.ProcessedVariable(
--> 165                     self.model.variables[key], self, self._known_evals
    166                 )
    167 

~\Documents\PyBaMM\pybamm\solvers\processed_variable.py in __init__(self, base_variable, solution, known_evals)
     66             if len(solution.t) == 1:
     67                 raise pybamm.SolverError(
---> 68                     "Solution time vector must have length > 1. Check whether "
     69                     "simulation terminated too early."
     70                 )

SolverError: Solution time vector must have length > 1. Check whether simulation terminated too early.

gyouhoc, 2020-10-27:
I think I found the issue. This is a silly mistake. Sorry for this. One of the parameters is not well defined. 

thx

closed 2020-10-27 by gyouhoc