pybamm-team/PyBaMM #1221: Optionally convert variable to casadi format before evaluating

tinosulzer, 2020-10-30:
Right now, for fast models such as the SPM, solving the model takes 2 ms but evaluating the voltage takes 160 ms. 

Under the right combination of conditions, it might be faster to convert a variable to casadi before evaluating it. The converted variable could be saved to a dictionary such as `model.casadi_variables` which would then be available to the next solution. This could also bypass the need for doing `y.full()` after solving to convert from casadi format to numpy format (`y.full()` takes as much time as doing the integration itself, 0.5 ms).

Could also try vectorizing as suggested in #788

TomTranter, 2020-10-31:
Would this work best for the volume averaged properties? If this were a CFD solver you could define monitoring points etc in the mesh to keep an eye on certain variables - not sure wee need to go that far and we're not monitoring convergence but I really like the idea of getting a few bits of key info on the fly - would help me solve the jellyroll problem a bit faster

tinosulzer, 2020-11-01:
I don't think volume-averaged or not makes a difference. There is no way to evaluate at just a single point without evaluating the whole thing first. (In general) the volume-averaged variables are the explicit integral (i.e. the dot product of a vector of ones and the variable) so you have to evaluate the full distribution before you can get the volume-averaged variable.

PR #1316 (2020-12-31): Issue 1221 convert to casadi
# Description

Convert variables to casadi for faster post-processing.
The conversion is done the first time the variable is called, and the casadi function is then stored in the _model_ (in `model._variables_casadi`. Therefore, when doing repeated solutions of the same model (e.g. for parameter sweeps) the conversion is only performed the first time.

Fixes #1221

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order o [... cut]
closed 2020-12-31 by tinosulzer