pybamm-team/PyBaMM #1235: sparse mat-vec support for EvaluatorJax

martinjrobins, 2020-11-07:
**Summary**

currently `pybamm.EvaluatorJax` does not support sparse matrices, because jax does not support them. It just replaces all sparse mats in expression tree with dense ones. However, you can implement a sparse mat-vec multiplication in jax easily enough (see https://gist.github.com/romanodev/34a471e8914989dbedee85febdbe7c77), so I propose replacing all sparse mat-vec mults with this, should be a lot faster.

 **Motivation**

a sparse mat-vec mult will be much faster than the equivalent dense mat-vec

martinjrobins, 2020-11-07:
anyone know if this will work for the DFN model? I had a brief play with it, and found a place in the expression tree where a constant sparse matrix is multiplied by a dense matrix (with entries depending on the state vector). It could be that I'm doing something wrong, but if anyone knows if there is definitely a matrix-matrix multiply in the DFN model, or any others, let me know!

tinosulzer, 2020-11-08:
The only `matmul` operations in the DFN model should be matrix-vector, or vector-vector (i.e. dot product) for "integration". If there are some matrix-matrix matmuls I would be interested to know where they are.
There may be some elementwise multiplications that are matrix-matrix though

martinjrobins, 2020-11-08:
you are absolutely right, this was me making a mistake and adding a 1D vector with a 2D column vector, which gave a 2D matrix via broadcasting. Fixed now

PR #1242 (2020-11-15): Issue 1235 jax sparse mat vec
# Description

`EvaluatorJax` uses a custom COO sparse matrix class and implements mat-vec multiply, as an alternative to converting everything to dense matrices

Fixes #1235 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x]  [... cut]
closed 2020-11-15 by martinjrobins