pybamm-team/PyBaMM #1236: R of x gets passed dimensionless x

rtimms, 2020-11-09:
The functions `R_n_of_x` etc. get passed dimensionless `x` but this doesn't get scaled when passed to the `FunctionParameter` which depends on `x [m]`, e.g.

```python3
        inputs = {"Through-cell distance (x_n) [m]": x}
        return pybamm.FunctionParameter("Negative particle distribution in x", inputs)
```

should be 

```python3
        inputs = {"Through-cell distance (x_n) [m]": x * self.L_x}
        return pybamm.FunctionParameter("Negative particle distribution in x", inputs)
```

since thicknesses are scaled by `L_x`

rtimms, 2020-11-09:
actually, more consistent with other things (e.g. initial solid concentration) to write it as a function of dimensionless through-cell position, i.e.

```python3
        inputs = {"Dimensionless through-cell position (x_n)": x}
        return pybamm.FunctionParameter("Negative particle distribution in x", inputs)
```

tinosulzer, 2020-11-09:
Looking at this again, why do we separate the particle radius and its dependence in x into two separate parameters? Wouldn't it be clearer if it was always a single (dimensional input, dimensional output) parameter?

rtimms, 2020-11-09:
yeah that would be cleaner. i think we kept it separate so that we had a reference radius to use in the non-dimensionlisation, though i guess you just take the average for that

tinosulzer, 2020-11-09:
Yeah, or evaluate at x=0 and x=1 (might have issues with the average not being defined until discretisation)

closed 2020-11-13 by rtimms