pybamm-team/PyBaMM #1243: Add JAX citation for the JAX solvers

tinosulzer, 2020-11-13:


tinosulzer, 2020-11-13:
Also add default ref to numpy paper (?)

@article{harris2020array,
  title={Array programming with NumPy},
  author={Harris, Charles R and Millman, K Jarrod and van der Walt, St{\'e}fan J and Gommers, Ralf and Virtanen, Pauli and Cournapeau, David and Wieser, Eric and Taylor, Julian and Berg, Sebastian and Smith, Nathaniel J and others},
  journal={Nature},
  volume={585},
  number={7825},
  pages={357--362},
  year={2020},
  publisher={Nature Publishing Group}
}

martinjrobins, 2020-11-13:
Re JAX, the developers ask you to cite the github repo itself (they are still working on a paper):
```
@software{jax2018github,
  author = {James Bradbury and Roy Frostig and Peter Hawkins and Matthew James Johnson and Chris Leary and Dougal Maclaurin and Skye Wanderman-Milne},
  title = {{JAX}: composable transformations of {P}ython+{N}um{P}y programs},
  url = {http://github.com/google/jax},
  version = {0.2.5},
  year = {2018},
}
```

PR #1258 (2020-11-23): Fix graphviz version for windows
# Description

Change version for graphviz for windows
also fixes #1243 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: [... cut]
closed 2020-11-23 by tinosulzer