pybamm-team/PyBaMM #1245: Increase in a active material during aging

optibikash, 2020-11-16:
I tried running cycling_ageing_yang.py for 50 cycles. I plotted the following quantities:
        "X-averaged negative electrode active material volume fraction",
        "X-averaged separator active material volume fraction",
        "X-averaged positive electrode active material volume fraction",
Each plot gave an increasing trend. AS per literature, with an increase in the number of cycles, the active material reduces which is one of the causes for degradation of battery or capacity fading.
All other values like terminal voltage and the dicharge capacity looked fine apart from the increase in the active material.

**To Reproduce**
Steps to reproduce the behavior:
Run the cycling_ageing_yang.py by changing the number "*2" to "*50 "in line 15 and run it. Plot the above quantities and check for the trend.

**Expected behavior**
An increase in active material will be seen.

**Possible solution**
The capacity fade can be computed as a space-time integral of ionic flux. Basically fade is the function of side reaction flux. The capacity fade computed can be used to update the initial active material at each cycle.

optibikash, 2020-11-16:
I am open for discussion to solve the issue. To a matter of fact, I am also working to solve the issue. I feel that I know how to solve the issue.

tinosulzer, 2020-11-16:
The problem is line 42 of `base_porosity.py`

optibikash, 2020-11-18:
Thanks, Tino. For the decrease in the active material, I have a suggestion to look into the thesis of Marcello Torchio who was a developer of LIONSIMBA. In his thesis, I have read about the decrease in the active material volume fraction and he has tried to implement it. I have tried implementing the same in MATLAB and could successfully implement that. The thesis is available online via this link: https://iris.unipv.it/retrieve/handle/11571/1203372/184976/main.pdf
On page 148 or chapter "Health-aware charging protocols for Li-ion batteries", the process is described in detail.

optibikash, 2020-11-18:
> 
> 
> We discussed this this morning. The problem is line 42 of `base_porosity.py`, where when the porosity decreases we assume that the inactive volume fraction stays constant and active material fraction increases. Instead, active material volume fraction should stay constant, and inactive volume fraction should increase.
> Note that pore clogging does not cause active material fraction to decrease, it will stay constant. There needs to be a different mechanism for active material loss.

So should I consider the "X-averaged ..........    active material volume fraction" as the increase in inactive material volume fraction in my study?

optibikash, 2020-11-18:
> We discussed this this morning. The problem is line 42 of `base_porosity.py`, where when the porosity decreases we assume that the inactive volume fraction stays constant and active material [...]

So probably what I have been plotting as "X-averaged negative electrode active material volume fraction", "X-averaged separator active material volume fraction", and "X-averaged positive electrode active material volume fraction", was actually the inactive material. Please correct me if I am wrong!

tinosulzer, 2020-11-18:
Yes, the increase you are seeing should be a change in inactive material, and active material volume fraction staying constant.
The way to fix this properly is to add a submodel for the active material volume fraction, which could either be constant or an ODE to model loss of active material

PR #1262 (2020-12-02): Issue 1245 active material
# Description

Set up active material models ready to model loss of active material
Fixes #1245 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [  [... cut]
closed 2020-12-02 by tinosulzer