pybamm-team/PyBaMM #1251: Diff of a ParameterFunction doesn't work when evaluated at a constant

brosaplanella, 2020-11-17:
**Describe the bug**
When defining a parameter as the `diff` of a `ParameterFunction` sometimes it returns 0. I think the issue is that the expression gets simplified first and, because the argument is a constant, we loose track of the differentiation variable. Then, when taking the derivative, the variable is not in the tree and therefore it returns 0.

**To Reproduce**
Steps to reproduce the behaviour:
1. Add a `diff` parameter to your model, for example
```python3
            def Uprime(sto, T):
                inputs = {"Positive particle stoichiometry": sto}
                Uprime_ref = pybamm.FunctionParameter(
                    "Positive electrode OCP [V]", inputs, diff_variable=sto
                )
                return Uprime_ref  / pot
```
2. Set the positive OCV from a function (for example a linear function).
3. Evaluate at `c_0 = param.c_p_init(1)`, the function returns zero.
4. Evaluate at `c_0 + 0 * pybamm.t`, the function now returns the right value.

tinosulzer, 2020-11-17:
Yeah that is exactly what is happening. Actually #1230 is going to make it even worse because it will automatically simplify `c_0 + 0 * pybamm.t` to `c_0` (but you can still force it with `pybamm.Addition(c_0, pybamm.Multiplication(0, pybamm.t))`.

It's not immediately obvious how to fix this. Ideally it only requires changing the logic in [parameter_values.py](https://github.com/pybamm-team/PyBaMM/blob/b892e5f59499c537d504de7fb3d56460623c0c83/pybamm/parameters/parameter_values.py#L544), rather than passing a flag around to avoid simplification

brosaplanella, 2020-11-17:
Can we get the derivative first and simplify afterwards? I tried finding how we could flip the order (found where the derivative is taken, but not where simplification happens). Is it obvious where it gets simplified so we can change the order?

tinosulzer, 2020-11-17:
Simplification happens as soon as the operator is created, e.g. in `Symbol.__add__`, so it can't be turned off as currently written

A hacky solution would be something like this
```python3
if symbol.diff_variable is not None:
    function = function_name(
        *[pybamm.Addition(child, pybamm.Multiplication(0, pybamm.t)) for child in new_children]
    )
else:
    function = function_name(*new_children)
```

brosaplanella, 2020-11-17:
Is the issue with `function` being simplified or with `diff_variable` being simplified? If the latter, could we just process `diff_variable` without simplifying it on creation?

tinosulzer, 2020-11-17:
I think it's the function

brosaplanella, 2020-11-17:
Ok, then I will implement the hacky method of the Multiplication by one, as you did with `c_e`.

tinosulzer, 2020-11-17:
You might need to do the adding 0 * pybamm.t for it to work. Otherwise even the multiplication will get simplified out later, e.g. `2 * pybamm.Multiplication(5, 1)` would realize that the whole thing is constant and return `Scalar(10)`

PR #1294 (2021-02-03): #1251 fix function parameter diff bug
# Description

Fixes #1251 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ pyt [... cut]
closed 2021-02-03 by tinosulzer