pybamm-team/PyBaMM #1253: Reformat DFN half-cell

brosaplanella, 2020-11-18:
Reformat the basic DFN half-cell so that the lithium counter is always on the left and the working electrode on the right, to be consistent with the standard DFN half-cell definitions of current and terminal voltage.

brosaplanella, 2020-11-18:
After playing around with it, I have noticed that this requires changing the definition of the geometry. Thinking of the long run as well (when we integrate it in the submodel structure), I am thinking of doing the following. Define a `half_cell_geometry` (similar to `battery_geometry`), which has a lithium electrode, a separator and a working (porous) electrode. When defining the geometry, we pass an option to define the working electrode and this sets the right parameters.

Is there a better way of doing it?

tinosulzer, 2020-11-18:
Can't think of a better way right now

tinosulzer, 2020-12-14:
As discussed today, the new format is much clearer and we should go ahead with it. Then, the half-cell models can use the submodel structure by passing the option "half-cell" or "full cell" to the submodels (this will require a little bit of submodel refactoring, but hopefully not too much).

PR #1282 (2020-12-22): Issue 1253 reformat dfn half cell
# Description

Reformatted the Basic DFN half-cell model. In the new setup, the lithium counter always sits on the left and the working electrode on the right. The new geometry has a `working electrode` and a `working particle`. 

Fixes #1253

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds fun [... cut]
closed 2020-12-22 by brosaplanella